cyanfish/naps2 #478: Canon DR-M140 pages get cut off when using SDK

cityvoice, 2024-10-29:
I'm having a problem with my canon scanners in using SDK. When I scan an A4 paper it cuts of the first ( or last ) 1cm of the page.
the same problem like: 
[](https://gitlab.com/sane-project/backends/-/issues/476#note_1735474004)

I'm using SDK to develop an electron app, I using browser to send a scan job to scanner with options:
{
            InputSource: 'ADF',
            Duplex: true,
            Resolution: 300,
            ScanRegion: 'A4',
}

it cut off the bottom 1cm of the page. But it works fine when using NAPS2 gui or command line tool NAPS2.Console.exe

so how to resolve this bug, thanks!

cyanfish, 2024-10-29:
Try "ScanRegion**s**" instead of "ScanRegion".

cityvoice, 2024-10-29:
> ScanRegion**s**

hi@cyanfish, very thanks for your replyment, I have tried "ScanRegions" instead of "ScanRegion", but the scan server throws 400 error.

the post data is :
`<?xml version="1.0" encoding="UTF-8" standalone="yes"?><scan:ScanSettings xmlns:scan="http://schemas.hp.com/imaging/escl/2011/05/03" xmlns:pwg="http://www.pwg.org/schemas/2010/12/sm" xmlns:dest="http://schemas.hp.com/imaging/httpdestination/2011/10/13"><pwg:InputSource>Feeder</pwg:InputSource><scan:Duplex>false</scan:Duplex><pwg:ScanRegions><pwg:ScanRegion pwg:MustHonor><pwg:Height>3507</pwg:Height><pwg:Width>2481</pwg:Width><pwg:XOffset>0</pwg:XOffset><pwg:YOffset>0</pwg:YOffset></pwg:ScanRegion></pwg:ScanRegions><pwg:Version>2</pwg:Version><scan:XResolution>300</scan:XResolution><scan:YResolution>300</scan:YResolution></scan:ScanSettings>`

this is the full error code:

> 
> scanner err AxiosError: Request failed with status code 400
>     at settle (file:///D:/projects/lianxing/pcclient/node_modules/.pnpm/axios@1.7.7/node_modules/axios/lib/core/settle.js:19:12)
>     at IncomingMessage.handleStreamEnd (file:///D:/projects/lianxing/pcclient/node_modules/.pnpm/axios@1.7.7/node_modules/axios/lib/adapters/http.js:599:11)
>     at IncomingMessage.emit (node:events:526:35)
>     at endReadableNT (node:internal/streams/readable:1408:12)
>     at process.processTicksAndRejections (node:internal/process/task_queues:82:21)
>     at Axios.request (file:///D:/projects/lianxing/pcclient/node_modules/.pnpm/axios@1.7.7/node_modules/axios/lib/core/Axios.js:45:41)
>     at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
>     at async Scanner.ScanJobs (file:///D:/projects/lianxing/pcclient/dist-electron/main/workers/scan.js:366:15)
>     at async startScan (file:///D:/projects/lianxing/pcclient/dist-electron/main/workers/scan.js:429:20) {
>   code: 'ERR_BAD_REQUEST',
>   config: {
>     transitional: {
>       silentJSONParsing: true,
>       forcedJSONParsing: true,
>       clarifyTimeoutError: false
>     },
>     adapter: [ 'xhr', 'http', 'fetch' ],
>     transformRequest: [ [Function: transformRequest] ],
>     transformResponse: [ [Function: transformResponse] ],
>     timeout: 0,
>     xsrfCookieName: 'XSRF-TOKEN',
>     xsrfHeaderName: 'X-XSRF-TOKEN',
>     maxContentLength: -1,
>     maxBodyLength: -1,
>     env: { FormData: [Function], Blob: [class Blob] },
>     validateStatus: [Function: validateStatus],
>     headers: Object [AxiosHeaders] {
>       Accept: 'application/json, text/plain, */*',
>       'Content-Type': 'application/x-www-form-urlencoded',
>       'User-Agent': 'axios/1.7.7',
>       'Content-Length': '648',
>       'Accept-Encoding': 'gzip, compress, deflate, br'
>     },
>     method: 'post',
>     data: '<?xml version="1.0" encoding="UTF-8" standalone="yes"?><scan:ScanSettings xmlns:scan="http://schemas.hp.com/imaging/escl/2011/05/03" xmlns:pwg="http://www.pwg.org/schemas/2010/12/sm" xmlns:dest="http://schemas.hp.com/imaging/httpdestination/2011/10/13"><pwg:InputSource>Feeder</pwg:InputSource><scan:Duplex>false</scan:Duplex><pwg:ScanRegions><pwg:ScanRegion pwg:MustHonor><pwg:Height>3507</pwg:Height><pwg:Width>2481</pwg:Width><pwg:XOffset>0</pwg:XOffset><pwg:YOffset>0</pwg:YOffset></pwg:ScanRegion></pwg:ScanRegions><pwg:Version>2</pwg:Version><scan:XResolution>300</scan:XResolution><scan:YResolution>300</scan:YResolution></scan:ScanSettings>',
>     url: 'http://127.0.0.1:9881/eSCL/ScanJobs'
>   },
>   request: <ref *1> ClientRequest {
>     _events: [Object: null prototype] {
>       abort: [Function (anonymous)],
>       aborted: [Function (anonymous)],
>       connect: [Function (anonymous)],
>       error: [Function (anonymous)],
>       socket: [Function (anonymous)],
>       timeout: [Function (anonymous)],
>       finish: [Function: requestOnFinish]
>     },
>     _eventsCount: 7,
>     _maxListeners: undefined,
>     outputData: [],
>     outputSize: 0,
>     writable: true,
>     destroyed: false,
>     _last: false,
>     chunkedEncoding: false,
>     shouldKeepAlive: false,
>     maxRequestsOnConnectionReached: false,
>     _defaultKeepAlive: true,
>     useChunkedEncodingByDefault: true,
[... 246 lines cut ...]
>       aborted: false,
>       timeoutCb: [Function: emitRequestTimeout],
>       upgradeOrConnect: false,
>       parser: null,
>       maxHeadersCount: null,
>       reusedSocket: false,
>       host: '127.0.0.1',
>       protocol: 'http:',
>       _redirectable: [Writable],
>       [Symbol(kCapture)]: false,
>       [Symbol(kBytesWritten)]: 0,
>       [Symbol(kNeedDrain)]: false,
>       [Symbol(corked)]: 0,
>       [Symbol(kOutHeaders)]: [Object: null prototype],
>       [Symbol(errored)]: null,
>       [Symbol(kHighWaterMark)]: 16384,
>       [Symbol(kRejectNonStandardBodyWrites)]: false,
>       [Symbol(kUniqueHeaders)]: null
>     },
>     data: ''
>   },
>   status: 400
> }

so how should i do? thanks

cyanfish, 2024-10-30:
Looks like a bug in the ESCL client library as it's generating invalid XML, I'm assuming you're using the code from the naps2-webscan sample? If so you can try commenting out lines 25 and 33 [here](https://github.com/cyanfish/naps2-webscan/blob/809e63f2c39669ff2cc224925ef7a1c82f87d351/NAPS2.WebScan.WebServer/wwwroot/lib/escl-sdk-ts/escl/serializeScanSetting.ts#L25) (the lines with "MustHonor").

cityvoice, 2024-11-13:
@cyanfish, thanks again, you are right,xml is invalid, change region to regions works.

cyanfish, 2025-04-19:
Closing as I've updated the naps2-webscan sample with the fix.

closed 2025-04-19 by cyanfish