cyanfish/naps2 #484: [Feature Request] Allow Enter Key to Confirm Crop

mwsmws22, 2024-11-16:
Very small feature request, but it would be nice if when cropping a page you could hit the "Enter" key to confirm the edit. Most photo editing apps works like this (e.g. snipping tool, photoshop). It's really hard to unwire my brain and remember to hit the "OK" button.

Currently, since the "Revert" button is the default selection in the crop tool, if you hit "Enter" it reverts all my edits. Annoying!

[image]

Maybe just change the default selected button to "OK"? Then "Enter" key should work.

Besides this tiny tiny thing, 10/10 app that I daily drive for my scanning needs. Thanks devs!

gmleuning, 2025-01-26:
I have been so annoyed by this. The problem is the button placement. With revert first, it pre-selects it. The easiest way to fix this is to change the button placement so OK is first. It took me a while to figure out, but then I used Deep Seek and after a few wrong turns, I found a solution that worked. 

I opened the folder with the downloaded repository in Visual Studio. In the Solution Explorer, I searched for ImageFormBase.cs, it is in naps2-master\NAPS2.Lib\EtoForms\Ui\ImageFormBase.cs

Once there, I changed 

 protected override void BuildLayout()
 {
     LayoutController.Content = L.Column(
         Overlay.Scale(),
         CreateControls(),
         L.Row(
             CreateExtraButtons(),
             C.Filler(),
             L.OkCancel(
                 C.OkButton(this, beforeClose: Apply),
                 C.CancelButton(this))
         )
     );
 }

to ->

protected override void BuildLayout()
{
    LayoutController.Content = L.Column(
        Overlay.Scale(),
        CreateControls(),
        L.Row(
            L.OkCancel(
                C.OkButton(this, beforeClose: Apply),
                C.CancelButton(this)
            ), 
            CreateExtraButtons(),
            C.Filler()
        )
    );
}

I am not sure if this will work for your version / machine, but I hope it points you in the right direction!

gmleuning, 2025-01-26:
I was really frustrated by this issue for a while. The problem is the button placement. With Revert first, it gets pre-selected by default. The easiest way to fix this is to change the button placement so OK comes first. It took me some time to figure out, this solution worked best for me.

Here’s what I did:

I opened the folder with the downloaded repository in Visual Studio.

In the Solution Explorer, I searched for ImageFormBase.cs. It’s located in:
> naps2-master\NAPS2.Lib\EtoForms\Ui\ImageFormBase.cs

This is the BuildLayout method:

```csharp
protected override void BuildLayout()
{
    LayoutController.Content = L.Column(
        Overlay.Scale(),
        CreateControls(),
        L.Row(
            CreateExtraButtons(),
            C.Filler(),
            L.OkCancel(
                C.OkButton(this, beforeClose: Apply),
                C.CancelButton(this))
        )
    );
}
```

And you can shift the buttons, making OK the default, by changing it to this:

```csharp

protected override void BuildLayout()
{
    LayoutController.Content = L.Column(
        Overlay.Scale(),
        CreateControls(),
        L.Row(
            L.OkCancel(
                C.OkButton(this, beforeClose: Apply),
                C.CancelButton(this)
            ), 
            CreateExtraButtons(),
            C.Filler()
        )
    );
}
```

This change ensures the OK button is pre-selected instead of the Revert button. I’m not sure if this will work for your specific version or setup, but I hope it points you in the right direction! And if you're not familiar with modifying in Visual Studio, I recommend Deep Seek, I had no idea where to start and it got me through with a little trial and error!

cyanfish, 2025-02-02:
This should be fixed in [8.0b3](https://www.naps2.com/download).

closed 2025-02-02 by cyanfish