cyanfish/naps2 #487: NAPS 7.5.2 maximized window is shown incompletely when opened as maximized

TobiasKnauss, 2024-11-24:
**Describe the bug**
See screenshot below.
NAPS was opened as maximized (because it was maximized when it was closed before). The inner part of the window is however not adapted to the maximized size, and therefore partly hidden.

**To Reproduce**
1. Start NAPS
2. Maximize the window
3. Close NAPS
4. Start NAPS again

**Expected behavior**
The GUI elements in the window should all be visible

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 Home (23H2)
 - Version 7.5.2 (was also present in previous version(s).

githubuser181226, 2024-11-30:
Hi. The same thing happening for me. Using Win7 x64 and Win10 x64. Small but very annoying bug.

cyanfish, 2024-12-30:
This should be fixed in [7.5.3](https://www.naps2.com/download).

commit 6757bbac (2024-12-23): WinForms: Fix loading maximization state and layouting

Fixes #453
Fixes #487
closed 2024-12-30 by cyanfish