cyanfish/naps2 #488: Split pages for all selected pages/images

Amshorko, 2024-11-26:
My request considers an ability to split (or possibly combine) all selected pages. Currently, this is possible under "Image" menu only with other features like crop, BW or sharpen with tickbox "Apply to all n selected images". I would like to be able to split all selected pages at the preselected point with just one click, ideally in the same way as in the mentioned features.

I usually work with pre-scanned PDFs and NAPS2 allows me to adjust and OCR those documents. If I desire to split pages that are double scanned, I have to do it manually page by page or find a different solution. By adding this feature, it wouldn't solve a problem for all situations, simply because there might be a need for different splitting point at some cases, however for homogenous documents, this is actually solid solution.

source124, 2025-01-21:
Additionally, allow manga mode (put the right half first), please. This is very tiring to do manually on scans. Thanks

AFulgens, 2025-03-16:
FWIW: Until split becomes available, I resort onto duplicating / quadrouplicating the scanned page and cropping individually 🤷‍♂️

cyanfish, 2025-03-29:
This is added in [8.1.2](https://www.naps2.com/download).

closed 2025-03-29 by cyanfish