cyanfish/naps2 #492: The worker process crashed. Grpc.Core.RpcException:

akpmohan07, 2024-12-01:
**Describe the bug**
`cat ~/.config/naps2/errorlog.txt`
A clear and concise description of what the bug is.
`2024-12-01 17:18:03.1883 50837 The worker process crashed. Grpc.Core.RpcException: Status(StatusCode="Unavailable", Detail="failed to connect to all addresses")
   at async Task<bool> GrpcDotNetNamedPipes.Internal.MessageReader<TMessage>.MoveNext(CancellationToken cancellationToken)
   at async Task NAPS2.Remoting.Worker.WorkerServiceAdapter.Scan(ScanningContext scanningContext, ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, string> imageCallback)
2024-12-01 17:20:03.8597 50837 The worker process crashed. Grpc.Core.RpcException: Status(StatusCode="Unavailable", Detail="failed to connect to all addresses")
   at async Task<bool> GrpcDotNetNamedPipes.Internal.MessageReader<TMessage>.MoveNext(CancellationToken cancellationToken)
   at async Task NAPS2.Remoting.Worker.WorkerServiceAdapter.Scan(ScanningContext scanningContext, ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, string> imageCallback)`

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10, macOS 12, Ubuntu 22.04]
 - Version [e.g. 7.1.0]

**Additional context**
Add any other context about the problem here.

[image]

MeneDev, 2025-02-11:
Just tried with NAPS2 7.5.3+46c5b271e7ea10fdec210c74e564d923f141d5c5

When I run `naps2.console --listdevices --driver sane`

`cat ~/.config/naps2/errorlog.txt` contains

```
2025-02-11 11:46:54.6627 43205 An error occurred that caused the console application to close. NAPS2.Scan.Exceptions.ScanDriverUnknownException: Der Arbeitsprozess ist abgestürzt.
 ---> Grpc.Core.RpcException: Status(StatusCode="Unavailable", Detail="failed to connect to all addresses")
   at async Task<bool> GrpcDotNetNamedPipes.Internal.MessageReader<TMessage>.MoveNext(CancellationToken cancellationToken)
   at async Task NAPS2.Remoting.Worker.WorkerServiceAdapter.GetDevices(ScanOptions options, CancellationToken cancelToken, Action<ScanDevice> callback)
   Exception_EndOfInnerExceptionStack
   at NAPS2.Remoting.Worker.WorkerServiceAdapter.GetDevices(ScanOptions options, CancellationToken cancelToken, Action`1 callback)
   at NAPS2.Scan.Internal.WorkerScanBridge.GetDevices(ScanOptions options, CancellationToken cancelToken, Action`1 callback)
   at NAPS2.Scan.ScanController.<>c__DisplayClass12_0.<<GetDevices>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at NAPS2.Util.AsyncProducers.<>c__DisplayClass2_0`1.<<RunProducer>b__0>d.MoveNext()
   at async IAsyncEnumerable<T> NAPS2.Util.AsyncSink<T>.AsAsyncEnumerable()+MoveNext()
   at async Task NAPS2.Automation.AutomatedScanning.ListDevices() x 2
   at async Task NAPS2.Automation.AutomatedScanning.Execute()
```

arvedviehweger, 2025-05-04:
I have the exact same issue on a Mac running macOS 15.4.1 sequoia with the latest version of NAPS2 (8.1.4).

`2025-05-04 10:13:07.4000 94838 The worker process crashed. Grpc.Core.RpcException: Status(StatusCode="Unavailable", Detail="failed to connect to all addresses")
   at async Task<bool> GrpcDotNetNamedPipes.Internal.MessageReader<TMessage>.MoveNext(CancellationToken cancellationToken)
   at async Task NAPS2.Remoting.Worker.WorkerServiceAdapter.Scan(ScanningContext scanningContext, ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, string> imageCallback)`

This only happens when using a Scanner with SANE. The same scanner works fine on the command line with scanimage, but it throws this error with NAPS2. I can hear the scanner scanning with NAPS2 but the app comes up with this error message and never shows the actual scanned image. 

If you need any more details or logs please let me know.

arvedviehweger, 2025-05-12:
After downloading the source code and building the project myself I managed to find the issue.

In line 50-51 in NAPS2.Sdk/Scan/Internal/WorkerScanBridge.cs

`using var ctx = _scanningContext.CreateWorker(_workerType)!;
await ctx.Service.Scan(_scanningContext, options, cancelToken, scanEvents,
 (image, tempPath) => { callback(image, new PostProcessingContext { TempPath = tempPath }); });`

is where the issue lies. Turns out the scanning request is being executed too early whilst the creation of the worker is not done yet. Introducing a small delay fixes the issue on my side and the worker no longer crashes.

`using var ctx = _scanningContext.CreateWorker(_workerType)!;
        await Task.Delay(500); // wait for worker to be ready
        await ctx.Service.Scan(_scanningContext, options, cancelToken, scanEvents,
            (image, tempPath) => { callback(image, new PostProcessingContext { TempPath = tempPath }); });`

There is probably a more elegant solution to this problem but my C#-skills are very limited. I hope this helps in properly fixing the issue. If anyone wants to try a build please let me know and I will upload it.

ss4max, 2025-05-20:
> After downloading the source code and building the project myself I managed to find the issue.
> 
> In line 50-51 in NAPS2.Sdk/Scan/Internal/WorkerScanBridge.cs
> 
> `using var ctx = _scanningContext.CreateWorker(_workerType)!;` `await ctx.Service.Scan(_scanningContext, options, cancelToken, scanEvents, (image, tempPath) => { callback(image, new PostProcessingContext { TempPath = tempPath }); });`
> 
> is where the issue lies. Turns out the scanning request is being executed too early whilst the creation of the worker is not done yet. Introducing a small delay fixes the issue on my side and the worker no longer crashes.
> 
> `using var ctx = _scanningContext.CreateWorker(_workerType)!;` `await Task.Delay(500); // wait for worker to be ready` `await ctx.Service.Scan(_scanningContext, options, cancelToken, scanEvents, (image, tempPath) => { callback(image, new PostProcessingContext { TempPath = tempPath }); });`
> 
> There is probably a more elegant solution to this problem but my C#-skills are very limited. I hope this helps in properly fixing the issue. [@cyanfish](https://github.com/cyanfish)
> 
> If anyone wants to try a build please let me know and I will upload it.

May I please try the build? I am having the same issue. Thanks!

arvedviehweger, 2025-05-20:
> May I please try the build? I am having the same issue. Thanks!

Sure!

Here it is: https://www.dropbox.com/scl/fi/l7fwklu1orlclonkhdi8m/NAPS2.dmg?rlkey=8tfyo1u4i1429ki1rf3htu6mi&st=1i0d5cnx&dl=0

ss4max, 2025-05-22:
> Sure!
>
> Here it is: https://www.dropbox.com/scl/fi/l7fwklu1orlclonkhdi8m/NAPS2.dmg?rlkey=8tfyo1u4i1429ki1rf3htu6mi&st=1i0d5cnx&dl=0

This works for me, thanks!

InfiniteTape, 2025-08-15:
> Sure!
>
> Here it is: https://www.dropbox.com/scl/fi/l7fwklu1orlclonkhdi8m/NAPS2.dmg?rlkey=8tfyo1u4i1429ki1rf3htu6mi&st=1i0d5cnx&dl=0

Worked for me as well. Thanks!

cyanfish, 2025-08-31:
This should be fixed in [8.2.1](https://www.naps2.com/download).

The issue is related to notarization (on newer macOS versions), I'm pretty sure the delay wasn't what fixed it - any debug build would have worked since it isn't notarized.

closed 2025-08-31 by cyanfish