cyanfish/naps2 #494: Memory Leak?

aidan-gibson, 2024-12-05:
**Describe the bug**
[image]
Kept using more and more memory until I force quit the process. Makes no sense, was a single page scan.

**To Reproduce**
Steps to reproduce the behavior:
Hit scan.
Scanner scans the page physically.
Memory usage climbs higher and higher.

**Desktop (please complete the following information):**
 - OS: macOS 15.1.1
 - Version 7.5.2

**Additional context**
EPSON ES-500W
Apple Driver (scans fine in image capture)
[image]
[image]

aidan-gibson, 2024-12-05:
[image]
What the GUI is showing while it's leaking.

I tried again with Maximum quality disabled, same result.

aidan-gibson, 2024-12-05:
Tried again with default settings (200dpi), same result.

However! The SANE version works.

cyanfish, 2025-01-31:
Can you try with 7.5.3?

aidan-gibson, 2025-02-01:
Works! Thanks dude

closed 2025-02-01 by aidan-gibson