cyanfish/naps2 #502: Duplex scan creating duplicate copies

RonMcKay, 2024-12-15:
**Describe the bug**
When I am scanning in duplex mode, the first page is getting scanned. Then the progress bar hangs at the second page for a while. After some time the scan is finished and a number of copies of the first scanned page are created. Sometimes I end up with up to 128 scans of the first page. Although, the exact number of duplicate copies is not always reproducible.

**To Reproduce**
Steps to reproduce the behavior:
1. Scan with the following settings
[image][image]
2. First page is duplicated many times
[image]

**Expected behavior**
The pages should be scanned normally.

**Screenshots**
See above.

**Desktop (please complete the following information):**
 - OS: macOS Sonoma 14.6.1
 - Version 7.5.2.0

**Additional context**
The problem occurs regardless of whether I scan in grayscale or color. Scanning with the Mac System interface of the scanner works fine, so I suspect the bug to be in NAPS2.

muescha, 2024-12-20:
# Hardware

- Brother MFC-7230 via USB
- installed scanner driver from brother home page

# Desktop (please complete the following information):

OS: macOS Sonoma 14.5 (23F79)
Version 7.5.2.0

# Problem

## Scanning with ADF

- All pages are scanned. after first page I get around 58x the same first page as image

## Scanning on flatbed

After double click on the profile I get immidiatly an error while scanning with flatbed:

`The operation couldn't be completed. (com.apple.imagecapture error
4294967249.)`

[image]

muescha, 2024-12-20:
Installed the 8.0.1 Beta:
- Flatbed: scan with no error
- AFD: has the same error: multiple first pages

Installed the 7.5.1:
- Flatbed: scan works as expected
- ADF: scan works as expected

muescha, 2024-12-21:
here is the error log:

<details>
<summary>Error Log</summary>

```text
2024-12-20 18:06:10.1277 68447 An error occurred that caused the worker task to terminate. System.InvalidOperationException: Response stream has already been completed.
   at Task GrpcDotNetNamedPipes.Internal.ResponseStreamWriterImpl<T>.WriteAsync(T message)
   at void System.Threading.Tasks.ContinuationResultTaskFromTask<TResult>.InnerInvoke()
   at void System.Threading.Tasks.Task.ExecuteWithThreadLocal(ref Task, Thread)
2024-12-20 18:06:10.5078 68447 An error occurred that caused the worker task to terminate. System.InvalidOperationException: Response stream has already been completed.
   at Task GrpcDotNetNamedPipes.Internal.ResponseStreamWriterImpl<T>.WriteAsync(T message)
   at void System.Threading.Tasks.ContinuationResultTaskFromTask<TResult>.InnerInvoke()
   at void System.Threading.Tasks.Task.ExecuteWithThreadLocal(ref Task, Thread)
2024-12-20 18:06:10.5078 68447 An error occurred that caused the worker task to terminate. System.InvalidOperationException: Response stream has already been completed.
   at Task GrpcDotNetNamedPipes.Internal.ResponseStreamWriterImpl<T>.WriteAsync(T message)
   at void System.Threading.Tasks.ContinuationResultTaskFromTask<TResult>.InnerInvoke()
   at void System.Threading.Tasks.Task.ExecuteWithThreadLocal(ref Task, Thread)
2024-12-20 18:14:46.3628 69496 Marshalling managed exception System.InvalidOperationException: MoreThanOneElement
   at void System.Linq.ThrowHelper.ThrowMoreThanOneElementException()
   at TSource System.Linq.Enumerable.TryGetSingle<TSource>(IEnumerable<TSource>, out bool)
   at TSource System.Linq.Enumerable.SingleOrDefault<TSource>(IEnumerable<TSource>)
   at ScanProfile NAPS2.EtoForms.Ui.ProfilesForm.get_SelectedProfile()
   at void NAPS2.EtoForms.Ui.ProfilesForm.ContextMenuOpening(object sender, EventArgs e)
   at void Eto.PropertyStore.TriggerEvent<T>(object key, object sender, T args)
   at void Eto.Forms.ContextMenu.OnOpening(EventArgs e)
   at void Eto.Forms.ContextMenu+Callback.OnOpening(ContextMenu widget, EventArgs e)
   at void Eto.Mac.Forms.Menu.ContextHandler.MenuWillOpen(NSMenu menu)
   at object InvokeStub_ContextHandler.MenuWillOpen(object, Span<object>)
   at object System.Reflection.MethodBaseInvoker.InvokeWithOneArg(object, BindingFlags, Binder, object[], CultureInfo)
   at object ObjCRuntime.Runtime.InvokeMethod(MethodBase, object, IntPtr) x 2
   at MonoObject* ObjCRuntime.Runtime.bridge_runtime_invoke_method(MonoObject* method, MonoObject* instance, IntPtr parameters, IntPtr* exception_gchandle)
2024-12-20 18:14:48.0753 69496 Marshalling managed exception System.InvalidOperationException: MoreThanOneElement
   at void System.Linq.ThrowHelper.ThrowMoreThanOneElementException()
   at TSource System.Linq.Enumerable.TryGetSingle<TSource>(IEnumerable<TSource>, out bool)
   at TSource System.Linq.Enumerable.SingleOrDefault<TSource>(IEnumerable<TSource>)
   at ScanProfile NAPS2.EtoForms.Ui.ProfilesForm.get_SelectedProfile()
   at void NAPS2.EtoForms.Ui.ProfilesForm.DoCopy()
   at NAPS2.EtoForms.ActionCommand(Action action)+(object sender, EventArgs args) => { }
   at void Eto.Forms.Command.OnExecuted(EventArgs e)
   at void Eto.Forms.Command.Execute()
   at void Eto.Forms.Command.System.Windows.Input.ICommand.Execute(object parameter)
   at void Eto.PropertyStore+CommandWrapper.Command_Execute(object sender, EventArgs e)
   at void Eto.Forms.MenuItem.OnClick(EventArgs e)
   at void Eto.Forms.MenuItem+Callback.OnClick(MenuItem widget, EventArgs e)
   at void Eto.Mac.Forms.Menu.ButtonMenuItemHandler<TWidget, TCallback>.Activate()
   at void Eto.Mac.Forms.Menu.MenuActionHandler.Activate(NSObject sender)
   at object InvokeStub_MenuActionHandler.Activate(object, Span<object>)
   at object System.Reflection.MethodBaseInvoker.InvokeWithOneArg(object, BindingFlags, Binder, object[], CultureInfo)
   at object ObjCRuntime.Runtime.InvokeMethod(MethodBase, object, IntPtr) x 2
   at MonoObject* ObjCRuntime.Runtime.bridge_runtime_invoke_method(MonoObject* method, MonoObject* instance, IntPtr parameters, IntPtr* exception_gchandle)
2024-12-20 19:09:04.6431 76931 Error running OCR System.Xml.XmlException: Xml_MissingRoot
   at void System.Xml.XmlTextReaderImpl.Throw(Exception)
   at void System.Xml.XmlTextReaderImpl.ThrowWithoutLineInfo(string)
   at bool System.Xml.XmlTextReaderImpl.ParseDocumentContent()
   at XDocument System.Xml.Linq.XDocument.Load(XmlReader, LoadOptions) x 3
   at async Task<OcrResult> NAPS2.Ocr.TesseractOcrEngine.ProcessImage(ScanningContext scanningContext, string imagePath, OcrParams ocrParams, CancellationToken cancelToken)
2024-12-20 19:09:10.6271 76931 Error running OCR System.Xml.XmlException: Xml_MissingRoot
   at void System.Xml.XmlTextReaderImpl.Throw(Exception)
   at void System.Xml.XmlTextReaderImpl.ThrowWithoutLineInfo(string)
   at bool System.Xml.XmlTextReaderImpl.ParseDocumentContent()
   at XDocument System.Xml.Linq.XDocument.Load(XmlReader, LoadOptions) x 3
   at async Task<OcrResult> NAPS2.Ocr.TesseractOcrEngine.ProcessImage(ScanningContext scanningContext, string imagePath, OcrParams ocrParams, CancellationToken cancelToken)
2024-12-20 19:09:18.7750 76931 Error running OCR System.Xml.XmlException: Xml_MissingRoot
   at void System.Xml.XmlTextReaderImpl.Throw(Exception)
   at void System.Xml.XmlTextReaderImpl.ThrowWithoutLineInfo(string)
   at bool System.Xml.XmlTextReaderImpl.ParseDocumentContent()
[... cut ...]
2024-12-20 19:10:16.5054 76931 Error running OCR System.Xml.XmlException: Xml_MissingRoot
   at void System.Xml.XmlTextReaderImpl.Throw(Exception)
   at void System.Xml.XmlTextReaderImpl.ThrowWithoutLineInfo(string)
   at bool System.Xml.XmlTextReaderImpl.ParseDocumentContent()
   at XDocument System.Xml.Linq.XDocument.Load(XmlReader, LoadOptions) x 3
   at async Task<OcrResult> NAPS2.Ocr.TesseractOcrEngine.ProcessImage(ScanningContext scanningContext, string imagePath, OcrParams ocrParams, CancellationToken cancelToken)

```

</details>

RonMcKay, 2024-12-25:
7.5.1 works also fine for me.

cyanfish, 2024-12-30:
This should be fixed in [7.5.3](https://www.naps2.com/download).

closed 2024-12-30 by cyanfish

RonMcKay, 2024-12-31:
I can confirm.
7.5.3 works for me, thanks! 🙂