cyanfish/naps2 #503: Fedora Linux, Process segfaults

jwrober, 2024-12-16:
**Describe the bug**
Segfault on Fedora 41

**To Reproduce**
Steps to reproduce the behavior:
1. Install Fedora 41 on your system
2. Download and install either 7.5.2 or 8x beta
3. Try to run
4. See error

```
$ naps2

Unhandled exception. System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation.
 ---> GLib.GException: Unrecognized image file format
   at Gdk.PixbufLoader.Write(Byte[] buf, UInt64 count)
   at Gdk.PixbufLoader.Write(Byte[] bytes, UInt32 count)
   at Gdk.PixbufLoader.LoadFromStream(Stream input)
   at Gdk.PixbufLoader.InitFromStream(Stream stream)
   at Gdk.PixbufLoader..ctor(Stream stream)
   at Gdk.Pixbuf..ctor(Stream stream)
   at Eto.GtkSharp.Drawing.IconHandler.Create(Stream stream)
   at Eto.Drawing.Icon..ctor(Stream stream)
   at NAPS2.EtoForms.EtoExtensions.ToEtoIcon(Byte[] bytes)
   at NAPS2.EtoForms.Ui.DesktopForm.BuildLayout()
   at NAPS2.EtoForms.EtoFormBase.OnPreLoad(EventArgs e)
   at Eto.Forms.Form.Show()
   at Eto.Forms.Application.<>c__DisplayClass63_0.<Run>b__0(Object sender, EventArgs e)
   at Eto.PropertyStore.TriggerEvent[T](Object key, Object sender, T args)
   at Eto.Forms.Application.OnInitialized(EventArgs e)
   at Eto.Forms.Application.Callback.OnInitialized(Application widget, EventArgs e)
   at Eto.GtkSharp.Forms.ApplicationHandler.Run()
   at Eto.Forms.Application.Run(Form mainForm)
   at NAPS2.EntryPoints.GuiEntryPoint.Run(String[] args, Module imageModule, Module guiModule)
   at NAPS2.EntryPoints.GtkEntryPoint.Run(String[] args)
   at System.RuntimeMethodHandle.InvokeMethod(Object, Void**, Signature, Boolean)
   at System.Reflection.MethodBaseInvoker.InterpretedInvoke_Method(Object obj, IntPtr* args)
   at System.Reflection.MethodBaseInvoker.InvokeDirectByRefWithFewArgs(Object, Span`1, BindingFlags)
   --- End of inner exception stack trace ---
   at System.Reflection.MethodBaseInvoker.InvokeDirectByRefWithFewArgs(Object, Span`1, BindingFlags)
   at System.Reflection.MethodBaseInvoker.InvokeWithOneArg(Object, BindingFlags, Binder, Object[], CultureInfo)
   at System.Reflection.RuntimeMethodInfo.Invoke(Object, BindingFlags, Binder, Object[], CultureInfo)
   at System.Reflection.MethodBase.Invoke(Object obj, Object[] parameters)
   at NAPS2.Program.Main(String[] args)
Aborted (core dumped)
```

**Expected behavior**
App opens. Worked on Fedora 40

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Fedora Linux 41, KDE Spin
 - Version 7.5.2-1

[image]

**Additional context**
None

sschuberth, 2024-12-20:
Getting the same thing, same Linux environment:

```
Unhandled exception. System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation.
 ---> Autofac.Core.DependencyResolutionException: An exception was thrown while activating NAPS2.EtoForms.Ui.GtkDesktopForm.
 ---> Autofac.Core.DependencyResolutionException: An exception was thrown while invoking the constructor 'Void .ctor(NAPS2.Config.Naps2Config, NAPS2.EtoForms.Desktop.DesktopKeyboardShortcuts, NAPS2.EtoForms.Notifications.NotificationManager, NAPS2.Util.CultureHelper, NAPS2.EtoForms.ColorScheme, NAPS2.Config.IProfileManager, NAPS2.Images.UiImageList, NAPS2.Images.ThumbnailController, NAPS2.Images.UiThumbnailProvider, NAPS2.EtoForms.Desktop.DesktopController, NAPS2.EtoForms.Desktop.IDesktopScanController, NAPS2.EtoForms.Desktop.ImageListActions, NAPS2.EtoForms.Widgets.ImageListViewBehavior, NAPS2.EtoForms.Desktop.DesktopFormProvider, NAPS2.EtoForms.Desktop.IDesktopSubFormController, System.Lazy`1[NAPS2.EtoForms.Ui.DesktopCommands], NAPS2.EtoForms.IDarkModeProvider, NAPS2.EtoForms.Desktop.Sidebar, NAPS2.EtoForms.IIconProvider)' on type 'GtkDesktopForm'.
 ---> GLib.GException: Unrecognized image file format
   at Gdk.PixbufLoader.Write(Byte[] buf, UInt64 count)
   at Gdk.PixbufLoader.Write(Byte[] bytes, UInt32 count)
   at Gdk.PixbufLoader.LoadFromStream(Stream input)
   at Gdk.PixbufLoader.InitFromStream(Stream stream)
   at Gdk.PixbufLoader..ctor(Stream stream)
   at Gdk.Pixbuf..ctor(Stream stream)
   at Eto.GtkSharp.Drawing.IconHandler.Create(Stream stream)
   at Eto.Drawing.Icon..ctor(Stream stream)
   at NAPS2.EtoForms.EtoExtensions.ToEtoIcon(Byte[] bytes)
   at NAPS2.EtoForms.Ui.DesktopForm..ctor(Naps2Config config, DesktopKeyboardShortcuts keyboardShortcuts, NotificationManager notificationManager, CultureHelper cultureHelper, ColorScheme colorScheme, IProfileManager profileManager, UiImageList imageList, ThumbnailController thumbnailController, UiThumbnailProvider thumbnailProvider, DesktopController desktopController, IDesktopScanController desktopScanController, ImageListActions imageListActions, ImageListViewBehavior imageListViewBehavior, DesktopFormProvider desktopFormProvider, IDesktopSubFormController desktopSubFormController, Lazy`1 commands, Sidebar sidebar, IIconProvider iconProvider)
   at NAPS2.EtoForms.Ui.GtkDesktopForm..ctor(Naps2Config config, DesktopKeyboardShortcuts keyboardShortcuts, NotificationManager notificationManager, CultureHelper cultureHelper, ColorScheme colorScheme, IProfileManager profileManager, UiImageList imageList, ThumbnailController thumbnailController, UiThumbnailProvider thumbnailProvider, DesktopController desktopController, IDesktopScanController desktopScanController, ImageListActions imageListActions, ImageListViewBehavior imageListViewBehavior, DesktopFormProvider desktopFormProvider, IDesktopSubFormController desktopSubFormController, Lazy`1 commands, IDarkModeProvider darkModeProvider, Sidebar sidebar, IIconProvider iconProvider)
   at lambda_method2(Closure, Object[])
   at Autofac.Core.Activators.Reflection.BoundConstructor.Instantiate()
   --- End of inner exception stack trace ---
   at Autofac.Core.Activators.Reflection.BoundConstructor.Instantiate()
   at Autofac.Core.Activators.Reflection.ReflectionActivator.<>c__DisplayClass14_0.<UseSingleConstructorActivation>b__0(ResolveRequestContext context, Action`1 next)
   at Autofac.Core.Resolving.Middleware.DelegateMiddleware.Execute(ResolveRequestContext context, Action`1 next)
   at Autofac.Core.Resolving.Pipeline.ResolvePipelineBuilder.<>c__DisplayClass14_0.<BuildPipeline>b__1(ResolveRequestContext context)
   at Autofac.Core.Resolving.Middleware.DisposalTrackingMiddleware.Execute(ResolveRequestContext context, Action`1 next)
   at Autofac.Core.Resolving.Pipeline.ResolvePipelineBuilder.<>c__DisplayClass14_0.<BuildPipeline>b__1(ResolveRequestContext context)
   at Autofac.Core.Resolving.Middleware.ActivatorErrorHandlingMiddleware.Execute(ResolveRequestContext context, Action`1 next)
   --- End of inner exception stack trace ---
   at Autofac.Core.Resolving.Middleware.ActivatorErrorHandlingMiddleware.Execute(ResolveRequestContext context, Action`1 next)
   at Autofac.Core.Resolving.Pipeline.ResolvePipelineBuilder.<>c__DisplayClass14_0.<BuildPipeline>b__1(ResolveRequestContext context)
   at Autofac.Core.Pipeline.ResolvePipeline.Invoke(ResolveRequestContext context)
   at Autofac.Core.Resolving.Middleware.RegistrationPipelineInvokeMiddleware.Execute(ResolveRequestContext context, Action`1 next)
   at Autofac.Core.Resolving.Pipeline.ResolvePipelineBuilder.<>c__DisplayClass14_0.<BuildPipeline>b__1(ResolveRequestContext context)
   at Autofac.Core.Resolving.Middleware.SharingMiddleware.Execute(ResolveRequestContext context, Action`1 next)
   at Autofac.Core.Resolving.Pipeline.ResolvePipelineBuilder.<>c__DisplayClass14_0.<BuildPipeline>b__1(ResolveRequestContext context)
   at Autofac.Core.Resolving.Pipeline.ResolvePipelineBuilder.<>c__DisplayClass14_0.<BuildPipeline>b__1(ResolveRequestContext context)
   at Autofac.Core.Resolving.Middleware.CircularDependencyDetectorMiddleware.Execute(ResolveRequestContext context, Action`1 next)
   at Autofac.Core.Resolving.Pipeline.ResolvePipelineBuilder.<>c__DisplayClass14_0.<BuildPipeline>b__1(ResolveRequestContext context)
   at Autofac.Core.Pipeline.ResolvePipeline.Invoke(ResolveRequestContext context)
[... cut ...]
   --- End of inner exception stack trace ---
   at System.Reflection.MethodBaseInvoker.InvokeDirectByRefWithFewArgs(Object, Span`1, BindingFlags)
   at System.Reflection.MethodBaseInvoker.InvokeWithOneArg(Object, BindingFlags, Binder, Object[], CultureInfo)
   at System.Reflection.RuntimeMethodInfo.Invoke(Object, BindingFlags, Binder, Object[], CultureInfo)
   at System.Reflection.MethodBase.Invoke(Object obj, Object[] parameters)
   at NAPS2.Program.Main(String[] args)
Aborted (core dumped)
```

rollonit, 2024-12-24:
I'm also having an identical issue (Fedora 41 and v8.0b1). Is there any chance it might have been caused by [this change](https://discourse.gnome.org/t/change-in-the-gdk-pixbuf-loaders-built-by-default-in-2-42-11/21845) in Pixbuf that removes support for a bunch of image types? I guess it was only merged in F41.

cyanfish, 2024-12-24:
@rollonit Yep that looks like it, thanks for the pointer.

cyanfish, 2024-12-30:
This should be fixed in [7.5.3](https://www.naps2.com/download).

closed 2024-12-30 by cyanfish

sschuberth, 2024-12-31:
> This should be fixed in [7.5.3](https://www.naps2.com/download).

Confirmed to be fixed, thanks a lot!