cyanfish/naps2 #505: .NET 6 has hit EOL

GGG-KILLER, 2024-12-18:
.NET 6 has hit EOL on November 12, 2024[^1] and will no longer be supported nor receive updates.
Recommended path is to upgrade either to .NET 8 (LTS) or .NET 9 (STS).

We've hit this while removing .NET 6 and 7 from nixpgks: https://github.com/NixOS/nixpkgs/issues/326335.

I'm aware that most of the projects were converted to .NET 8.0 on 61aebe6c8c033c2ea65e7082e1d2cc385c825077, however there are still a few that depend on .NET 6.0 and that's an issue for us as we want to completely remove it and it might result in NAPS2 being removed as well long-term.

[^1]: https://dotnet.microsoft.com/en-us/platform/support/policy/dotnet-core

cyanfish, 2024-12-30:
Can you try building again off `master`?

cyanfish, 2025-01-27:
Going to close this for now, feel free to comment if you have further issues and I'll reopen.

closed 2025-01-27 by cyanfish