cyanfish/naps2 #506: NAPS v. 7.5.2.0 Crash merging imported PDF files in one PDF file (Windows)

m4t3u-Dls, 2024-12-20:
**Describe the bug**
APP crash merging distinct imported PDF files in one new PDF unique file. Tested with one 12 pages, 12 MB PDF file + 5-6 pdf files with 3 pages and 1 MB files.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Import'
2. Select a bunch of PDF files.
3. Can use ORC or not, this does not change the result, maybe it could with small size PDF file.
4. Clik on 'Save PDF'

**Expected behavior**
Creating a new PDF unique file containing all imported PDF files.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 24H2 (26100.2605)

**Windows Event Log errors:**
```
`Depósito con errores , tipo 0
Nombre de evento: APPCRASH
Respuesta: No disponible
Identificador de archivo .cab: 0

Firma del problema:
P1: NAPS2.exe
P2: 7.5.2.0
P3: 817aa785
P4: pdfium.dll
P5: 123.0.6268.0
P6: 656c0b50
P7: c0000005
P8: 0000000000058ad5
P9: 
P10: 

Archivos adjuntos:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.9ee9cb6d-8535-43d3-a1de-0190ce1060f4.tmp.WERInternalMetadata.xml

Es posible que estos archivos estén disponibles aquí:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportQueue\AppCrash_NAPS2.exe_56e3b6d7414c18e4c5cf6028acd69ef2a4bac77_a434fa73_fef2201e-1e7e-4ccd-9f0d-c4308476d5b1

Símbolo de análisis: 
Nueva búsqueda de una solución: 0
Id. de informe: 21bc720a-e270-4fa2-9510-905e5bbbf188
Estado del informe: 100
Depósito con algoritmo hash: 
GUID de archivo .cab: 0`
```

```
`Nombre de aplicación con errores: NAPS2.exe, versión: 7.5.2.0, marca de tiempo: 0x817aa785
Nombre del módulo con errores: pdfium.dll, versión: 123.0.6268.0, marca de tiempo: 0x656c0b50
Código de excepción: 0xc0000005
Desplazamiento con errores: 0x0000000000058ad5
Id. de proceso con errores: 0x2B70
Tiempo de inicio de aplicación con errores: 0x1DB52B30181922A
Ruta de aplicación con errores: C:\Program Files\NAPS2\NAPS2.exe
Ruta de módulo con errores: C:\Program Files\NAPS2\lib\_win64\pdfium.dll
Id. de informe: 21bc720a-e270-4fa2-9510-905e5bbbf188
Nombre completo del paquete con errores: 
Id. de aplicación relacionado con el paquete con errores: `
```

```
`Aplicación: NAPS2.exe
Versión de Framework: v4.0.30319
Descripción: el proceso terminó debido a una excepción no controlada.
Información de la excepción: System.AccessViolationException
   en NAPS2.Pdf.Pdfium.PdfDocument.Save(System.IO.Stream)
   en NAPS2.Pdf.Pdfium.PdfDocument.Save(System.String)
   en NAPS2.Pdf.PdfExporter.MergePassthroughPages(System.IO.MemoryStream, OutputPathOrStream, System.Collections.Generic.List`1<PageExportState>, NAPS2.Pdf.PdfExportParams, NAPS2.Util.ProgressHandler)
   en NAPS2.Pdf.PdfExporter+<>c__DisplayClass8_0+<<Export>b__0>d.MoveNext()
   en System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   en System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   en System.Runtime.CompilerServices.AsyncMethodBuilderCore+MoveNextRunner.Run()
   en System.Threading.Tasks.AwaitTaskContinuation.RunOrScheduleAction(System.Action, Boolean, System.Threading.Tasks.Task ByRef)
   en System.Threading.Tasks.Task.FinishContinuations()
   en System.Threading.Tasks.Task`1[[System.__Canon, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]].TrySetResult(System.__Canon)
   en System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1[[System.__Canon, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]].SetResult(System.__Canon)
   en NAPS2.Threading.Pipeline+PipelineStep`1+<Run>d__10[[System.__Canon, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]].MoveNext()
   en System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   en System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   en System.Runtime.CompilerServices.AsyncMethodBuilderCore+MoveNextRunner.Run()
   en System.Threading.Tasks.AwaitTaskContinuation.RunOrScheduleAction(System.Action, Boolean, System.Threading.Tasks.Task ByRef)
   en System.Threading.Tasks.Task.FinishContinuations()
   en System.Threading.Tasks.Task`1[[System.Threading.Tasks.Dataflow.Internal.VoidResult, System.Threading.Tasks.Dataflow, Version=8.0.0.1, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a]].TrySetResult(System.Threading.Tasks.Dataflow.Internal.VoidResult)
   en System.Threading.Tasks.TaskCompletionSource`1[[System.Threading.Tasks.Dataflow.Internal.VoidResult, System.Threading.Tasks.Dataflow, Version=8.0.0.1, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a]].TrySetResult(System.Threading.Tasks.Dataflow.Internal.VoidResult)
   en System.Threading.Tasks.Dataflow.Internal.TargetCore`1[[System.__Canon, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]].CompleteBlockOncePossible()
   en System.Threading.Tasks.Task.Execute()
   en System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   en System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   en System.Threading.Tasks.Task.ExecuteWithThreadLocal(System.Threading.Tasks.Task ByRef)
   en System.Threading.Tasks.Task.ExecuteEntry(Boolean)
   en System.Threading.ThreadPoolWorkQueue.Dispatch()
`
```

[ReportQueue20241220.zip](https://github.com/user-attachments/files/18207879/ReportQueue20241220.zip)
[WER.9ee9cb6d-8535-43d3-a1de-0190ce1060f4.tmp.WERInternalMetadata20241220.zip](https://github.com/user-attachments/files/18207884/WER.9ee9cb6d-8535-43d3-a1de-0190ce1060f4.tmp.WERInternalMetadata20241220.zip)

bubbletroubles, 2025-02-22:
@cyanfish I can reproduce this on NAPS2 v 7.53 and 8.0b3. I've attached all the error logs I can find - if you'd like more, I'm happy to supply them.

```
Faulting application name: NAPS2.exe, version: 8.0.3.0, time stamp: 0x67200000
Faulting module name: coreclr.dll, version: 9.0.24.52809, time stamp: 0x672049fc
Exception code: 0xc0000005
Fault offset: 0x00000000002eea8b
Faulting process id: 0x21F4
Faulting application start time: 0x1DB84F4CFFFF8F0
Faulting application path: C:\Program Files\NAPS2\NAPS2.exe
Faulting module path: C:\Program Files\NAPS2\lib\coreclr.dll
Report Id: 103ec436-905c-485f-bcfd-76fbcc716856
Faulting package full name: 
Faulting package-relative application ID: 
```

```
Fault bucket 1389526189409383590, type 4
Event Name: APPCRASH
Response: Not available
Cab Id: 0

Problem signature:
P1: NAPS2.exe
P2: 8.0.3.0
P3: 67200000
P4: coreclr.dll
P5: 9.0.24.52809
P6: 672049fc
P7: c0000005
P8: 00000000002eea8b
P9: 
P10: 

Attached files:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.aa8c99a7-62d9-47c1-927e-8663cb087c7e.tmp.mdmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.dd12ca3b-57bc-44af-8c3f-912386c33c1e.tmp.WERInternalMetadata.xml
WPR_initiated_DiagTrackMiniLogger_OneTrace_User_Logger_20240808_1_EC_0_inject.etl
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.9d826429-11fb-402d-b3d5-7cfbdab8895e.tmp.etl
WPR_initiated_DiagTrackMiniLogger_WPR System Collector_inject.etl
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.8556e44c-4e0c-4e11-8eff-882e34048809.tmp.etl
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.08788428-07e4-4003-8a26-74934502e776.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.55015c04-889d-4482-98fa-bcad37ee01b2.tmp.txt
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.81b7df87-c566-4410-a7f5-49464f47d7ca.tmp.xml

These files may be available here:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_NAPS2.exe_42aa718caa30783493083576c5899d06baffb89_2c7dd08a_95aab25f-b01a-49ea-8458-a372e750a722

Analysis symbol: 
Rechecking for solution: 0
Report Id: 103ec436-905c-485f-bcfd-76fbcc716856
Report Status: 268435456
Hashed bucket: dc60d656ec72ad56234896b50dd934a6
Cab Guid: 0
```

```
Application: NAPS2.exe
CoreCLR Version: 9.0.24.52809
.NET Version: 9.0.0
Description: The process was terminated due to an unhandled exception.
Stack:
   at NAPS2.Pdf.Pdfium.PdfDocument.Save(System.IO.Stream)
   at NAPS2.Pdf.Pdfium.PdfDocument.Save(System.String)
   at NAPS2.ImportExport.OutputPathOrStream.SavePdfDoc(NAPS2.Pdf.Pdfium.PdfDocument)
   at NAPS2.Pdf.PdfExporter.MergePassthroughPages(System.IO.MemoryStream, NAPS2.ImportExport.OutputPathOrStream, System.Collections.Generic.List`1<PageExportState>, NAPS2.Pdf.PdfExportParams, NAPS2.Util.ProgressHandler)
   at NAPS2.Pdf.PdfExporter+<>c__DisplayClass8_0+<<Export>b__0>d.MoveNext()
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1+AsyncStateMachineBox`1[[System.Boolean, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[NAPS2.Pdf.PdfExporter+<>c__DisplayClass8_0+<<Export>b__0>d, NAPS2.Sdk, Version=8.0.3.0, Culture=neutral, PublicKeyToken=null]].ExecutionContextCallback(System.Object)
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1+AsyncStateMachineBox`1[[System.Boolean, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[NAPS2.Pdf.PdfExporter+<>c__DisplayClass8_0+<<Export>b__0>d, NAPS2.Sdk, Version=8.0.3.0, Culture=neutral, PublicKeyToken=null]].MoveNext(System.Threading.Thread)
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1+AsyncStateMachineBox`1[[System.Boolean, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[NAPS2.Pdf.PdfExporter+<>c__DisplayClass8_0+<<Export>b__0>d, NAPS2.Sdk, Version=8.0.3.0, Culture=neutral, PublicKeyToken=null]].MoveNext()
   at System.Threading.Tasks.AwaitTaskContinuation.RunOrScheduleAction(System.Runtime.CompilerServices.IAsyncStateMachineBox, Boolean)
   at System.Threading.Tasks.Task.RunContinuations(System.Object)
   at System.Threading.Tasks.Task`1[[System.__Canon, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].TrySetResult(System.__Canon)
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1[[System.__Canon, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].SetExistingTaskResult(System.Threading.Tasks.Task`1<System.__Canon>, System.__Canon)
   at NAPS2.Threading.Pipeline+PipelineStep`1+<Run>d__10[[System.__Canon, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].MoveNext()
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1+AsyncStateMachineBox`1[[System.__Canon, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[NAPS2.Threading.Pipeline+PipelineStep`1+<Run>d__10[[System.__Canon, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]], NAPS2.Sdk, Version=8.0.3.0, Culture=neutral, PublicKeyToken=null]].ExecutionContextCallback(System.Object)
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
[... cut ...]
   at System.Threading.Tasks.Dataflow.Internal.TargetCore`1[[System.__Canon, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].CompleteBlockOncePossible()
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(System.Threading.Thread, System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(System.Threading.Tasks.Task ByRef, System.Threading.Thread)
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading.PortableThreadPool+WorkerThread.WorkerThreadStart()
```

cyanfish, 2026-10-09:
Tracking in #918

closed 2026-10-09 by cyanfish