cyanfish/naps2 #515: SDK version 1.1.2  Error with Twain driver. "The selected scanner is offline".

bernhard-bernan, 2025-01-07:
### Discussed in https://github.com/cyanfish/naps2/discussions/504

<div type='discussions-op-text'>

<sup>Originally posted by **bernhard-bernan** December 17, 2024</sup>
Hello, Im using **SDK** version **1.1.2** and im getting error "The selected scanner is offline" when scanning page. This error doesnt happen with **NAPS2 7.5.2** desktop app. Im getting it only in SDK in my own project. This is pseudo code that kinda is similar to real code:

```
public class ScannerManager : IScannerManager
{
    private readonly ScanController _scanController;

    public ScannerManager()
    {
      var scanningContext = new ScanningContext(new ImageSharpImageContext());
      scanningContext.SetUpWin32Worker();
      _scanController = new ScanController(scanningContext);
      _scanController.PropagateErrors = false;
      _scanController.ScanEnd += OnScanEnd;
    }

    public async Task<List<byte[]>> ScanPagesBasedOnOptionsAsync(ScanOptions options)
    {
      var imageList = new List<byte[]>();

      await foreach (var img in _scanController.Scan(options))
      {
        var imgBytes = img.SaveToMemoryStream(ImageFileFormat.Jpeg).ToArray();
        imageList.Add(imgBytes);
      }
      return imageList;
    }

    public async Task<ScanDevice> FindTwainDeviceByNameAsync(string name)
    {
      var options = new ScanOptions() { Driver = Driver.Twain, };
      return await FindDeviceByNameAsync(name, options);
    }

    private async Task<ScanDevice> FindDeviceByNameAsync(string name, ScanOptions options)
    {
      var devices = await _scanController.GetDeviceList(options);

      foreach (var device in devices)
      {
        if (device.Name.Equals(name) && device.Driver.Equals(options.Driver))
        {
          return device;
        }
      }

      throw new ScannerException($"{ExceptionMessages.ScannerNotFound}:{name}");
    }

    private void OnScanEnd(object? sender, ScanEndEventArgs args)
    {
      if (args == null || args.Error == null)
      {
                return;
      }
      throw new ScannerException(args.Error.Message);
    }

}

public class ScannersService
{
  private readonly IScannerManager _scannerManager;

  public ScannersService(IScannerManager scannerManager)
  {
    _imageManager = imageManager;
  }

  public async Task ScanPage(ScannerRecord scannerRecord)
  {
    ScanDevice device = await _scannerManager.FindTwainDeviceByNameAsync(scannerRecord.Name);

    ScanOptions options = CreateScannerOptions(scannerRecord, device);
    try
    {

      List<byte[]> scannedImages = await _scannerManager.ScanPagesBasedOnOptionsAsync(options);
    }
    catch (Exception e)
    {
      throw new ScannerException(e.Message);
    }
  }

  private ScanOptions CreateScannerOptions(ScannerRecord scannerRecord, ScanDevice device) {
    // Some logic to get source type by record value(PaperSource - Auto, Flatbed, Feeder, Duplex;)
    PaperSource paperSource = GetPaperSourceTypeByRecord(scannerRecord);

    return new()
    {
      Device = device,
      PaperSource = paperSource,
      PageSize = NAPS2.Images.PageSize.A4,
      Dpi = scannerRecord.DpiQuality.Value,
      BitDepth = NAPS2.Images.BitDepth.Color,
      Driver = Driver.Twain,
    };
  }

}

```

Could it be related to how i set options, or that i find device before scanning ? I checked your code and to me it seems that you create new Device object instead of searching for device, could that be the problem ? :
```
    private async Task<bool> PopulateDevice(ScanProfile scanProfile, ScanOptions options)
    {
        // If a device wasn't specified, prompt the user to pick one
        if (string.IsNullOrEmpty(scanProfile.Device?.ID))
        {
            options.Device = (await _devicePrompt.PromptForDevice(options, false)).Device;
            if (options.Device == null)
            {
                return false;
            }

            // Persist the device in the profile if configured to do so
            if (_config.Get(c => c.AlwaysRememberDevice))
            {
                scanProfile.Device = ScanProfileDevice.FromScanDevice(options.Device);
                _profileManager.Save();
            }
        }
        else
        {
            options.Device = scanProfile.Device?.ToScanDevice(options.Driver);
        }

        return true;
    }
```
Any help appreciated!

</div>

cyanfish, 2025-01-08:
You probably need to set up a worker, see https://github.com/cyanfish/naps2/blob/master/NAPS2.Sdk.Samples/TwainSample.cs

cyanfish, 2025-01-08:
Do you happen to be running this in a Windows Service?

bernhard-bernan, 2025-01-08:
I mean i have it as desktop app created with .net electron, asp and angular so technically i do have embedded server that is wrapped into desktop app.

cyanfish, 2025-01-08:
Can you try with version 1.1.3?

bernhard-bernan, 2025-01-08:
Actually, at the start, I had version 1.1.3, but then I encountered not only this issue but also issues with some scanners that were directly connected to the PC via USB. So i downgraded to 1.1.2.

cyanfish, 2025-03-20:
I think 1.1.5 should fix the regression in 1.1.3.

closed 2025-06-08 by cyanfish