cyanfish/naps2 #516: "naps2-8.0b2-win-x64.zip" detected as malware by Microsoft Defender

tomasz1986, 2025-01-07:
**Describe the bug**

Microsoft Defender incorrectly flags the Windows installer from https://github.com/cyanfish/naps2/releases/tag/v8.0b2 as "Trojan:Script/Wacatac.B!ml".

**To Reproduce**

Steps to reproduce the behavior:

1. Try downloading https://github.com/cyanfish/naps2/releases/download/v8.0b2/naps2-8.0b2-win-x64.zip on a Windows 10/11 machine with the newest Microsoft Defender definition updates.
2. The downloaded file is immediately blocked by Microsoft Defender.

**Expected behavior**

The downloaded file should not be flagged by Microsoft Defender as malware.

**Desktop (please complete the following information):**

 - OS: Windows 11
 - Version 8.0b2

**Additional context**

I know that this is likely a false positive, but I'm posting this just in case. I have also submitted the file to Microsoft for revision. The submission is available at https://microsoft.com/en-us/wdsi/submission/438cf6d7-2b41-4459-9d8c-48524533d52a, but I think you need to have a Microsoft account in order to view it.

[image]

For the record, Microsoft Defender is not the only antivirus software that has issues with the file, which on Virus Total currently has a score of 15/66 (see https://www.virustotal.com/gui/file/5589df0eef6fca5ab39b0f91a00cd1ee5b3ad1bb320c6c4c04891279e5703e95).

[image]

borouhin, 2025-01-12:
To be more specific, it's not ZIP archive itself, but "NAPS2.Worker.exe" file inside it. It raises Defender Alarm avter v. 8.0b2 is installed via installer too. This file scores 27/71 on Virustotal currently, which IMO is rather alarming.
[image]

cyanfish, 2025-01-12:
The zip is down to 8 (from 15) as some of the false positives have been resolved which is progress at least.

tomasz1986, 2025-01-12:
I can also add that just today the detected "malware" in Windows Defender has changed from "Trojan:Script/Wacatac.B!ml" to "Virus:Win32/virut". Microsoft hasn't responded to my submission yet (and I somewhat doubt sure they ever will).

cyanfish, 2025-01-12:
I suspect the change from 8.0b1 to 8.0b2 is related to changing the .NET toolchain from 9RC2 to 9.0.101. Presumably there's been some change in the way the new version builds NAPS2 that is being flagged by the signature detection.

Ah yeah looks like others have similar issues:
https://github.com/dotnet/runtime/issues/110541
https://dylanbeattie.net/2024/12/09/dotnet-9-aot-zip-file-windows-defender-false-positives.html

cyanfish, 2025-01-12:
It also looks like I introduced a bug where the worker isn't signed properly (now fixed in https://github.com/cyanfish/naps2/commit/6ea12cf70807249465239b0c2eae3ea6ce45b933). While there's clearly still an issue with the detection, at least that should fix the problems for the next beta.

winman3000, 2025-01-16:
Same with the .MSI-File.

cyanfish, 2025-02-02:
This should be fixed in [8.0b3](https://www.naps2.com/download).

closed 2025-02-02 by cyanfish