cyanfish/naps2 #519: Duplicate profiles appearing in "New scan" window

mrsilver76, 2025-01-12:
I appear to have two profiles called "Documents" (both of which have the same settings) and a profile called "Photo" and "Photo (default)", both of which also have the same settings.

There seems to be no way in the UI to edit or delete these items. The obvious location would be "Profiles" but there is nothing in "Profile settings" or "Advanced".

[image]

I looked at manually editing the XML files to remove them - but neither `config.xml` or `profiles.xml` in `%appdata%\Roaming\naps2` contains anything that looks like this information.

Version: 7.5.3.0
OS: Windows 11 Home, 24H2, build 26100.2605

cyanfish, 2025-01-12:
These aren't part of NAPS2. Since you have "Use native UI" selected you're using the Windows scanning interface. Have a look at https://answers.microsoft.com/en-us/windows/forum/all/editing-windows-fax-scan-scan-profiles/72fac3e7-4bf0-401a-be22-321093ffea1f

mrsilver76, 2025-01-12:
Apologies and thank you! 

Can I make a suggestion which might avoid someone else asking this in the future?

Within the NAPS2 "Profile settings", would you consider changing "Use native UI" to something like "Use native Windows Scanner UI"? Even better, you could also include something that launches `control sticpl.cpl` - which takes you directly to the settings for that native UI.

Many thanks.