cyanfish/naps2 #528: Canon LiDE 20 - not correctly recognized/not working as it should according to available information.

Eiion, 2025-01-18:
**Describe the bug**
After creating the Profile an error message pops up.

**To Reproduce**
Steps to reproduce the behavior:
1. connect Scanner via USB to PC
2. install NAPS2
3. Click on 'Scan'
4. find scanner and adjust settings for it (adjust page size)
5. Click on 'Ok'
6. get error message popup that says (on a German language system)
```
---------------------------
ScanGear CS
---------------------------
Der Scanner ist nicht verfügbar, wenn er von einer anderen Anwendung oder einem anderen Benutzer verwendet wird.
---------------------------
OK   
---------------------------
```
(translation: The scanner isn't available, when it's in use by another application or another user.)

**Expected behavior**
Click "Ok" and the scanning will start. (According to the Installation instructions on your website [here](https://www.naps2.com/windows-scanning#installation)

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro 64bit
 - Version: 22H2

**Additional context**
As for the error message: There is no other user or application that is using the scanner. It's only connected to the system without previously any driver or software installed for the scanner, since none official driver or software is available anymore for many years already.

As for the incorrect recognition of the scanner:
Before deciding to go with NAPS2 I checked with the sane project page to see if my old scanner is compatible. See the project page device compatibility list [here](http://www.sane-project.org/sane-mfgs.html#Z-CANON). For the Canon LiDE 20 it says
```
CanoScan LiDE 20 | USB | 0x04a9/0x220d | Complete | Identical to CanoScan N670U | plustek(0.52) | sane-plustek
(0.52) | [sane-plustek](http://www.sane-project.org/man/sane-plustek.5.html)
```
Going by that piece of information I have to assume that the Canon LiDE 20 already has been successfully integrated into the software.

When creating the profile, after clicking "Choose device", none shows up for WIA or ESCL Drivers. For TWAIN, instead of the LiDE 20 the LiDE 25 and the LiDE 60 get listed. With that I can only assume that all three scanners are working with the same driver.
Selecting either of the two available scanners, both produce the error message.

The questions would be:
Why is the LiDE 20 not listed, was something done incorrectly (points 1-6) on my side or are the instructions missing a vital step?

Either way, with the LiDE 20 being listed as Completed in the backend part of the NAPS2 (i.e. the Sane Project), there appears to be a bug that causes the scanner not to be listed in device selection and after selecting either one of the two available an error message pops up.

funkerwolf, 2025-01-18:
This model is ancient junk, there will be guaranteed problems with it and the main one is that if there is a 64-bit driver for it, TWAIN itself is 32-broken in it. You need to use TWAINDSM for it. I'm writing this because I dealt with it myself and it was a big problem to make it work with CorelDRAW 64-bit version, but in 32 all started with the replacement with TWAINDSM.

cyanfish, 2025-01-18:
NAPS2 doesn't use SANE on Windows, only Linux and Mac. On Windows you'll need a manufacturer driver for it to work (except for newer network scanners that use ESCL).

Eiion, 2025-01-18:
@funkerwolf Thanks for evaluating my scanner.
I think you meant to say "that there is _not_ a 64-bit driver for it".
The video you have linked is in Russian language and the auto translation to English (or another language for that matter) is too bad to follow along with what the guy is saying. I might google TWAINDSM though.

@cyanfish Well, that's a piece of information I didn't see on the SANE (which lists NPAS2 as frontend for windows systems) or the NAPS2 website. Good to know, I guess.

I do have the original 32 bit TWAIN driver from [Canon's website](https://www.usa.canon.com/support/p/canoscan-lide-20?subtab=downloads-drivers). It's still available for downloading.

funkerwolf, 2025-01-18:
> [@funkerwolf](https://github.com/funkerwolf) Thanks for evaluating my scanner. I think you meant to say "that there is _not_ a 64-bit driver for it". The video you have linked is in Russian language [...]

Your case seems a bit different, but the model is similar, but the drivers are not compatible, it might be easier to switch to vuescan.

cyanfish, 2025-01-26:
Closing but created https://github.com/cyanfish/naps2/issues/547 as a tracking issue

closed 2025-01-26 by cyanfish