cyanfish/naps2 #529: Issue with the Flatpak package

FlawlessCasual17, 2025-01-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Be sure your operating system is Linux based, and is using KDE Plasma 6 as the desktop environment.
2. Install the flatpak version with: `flatpak install -y ./naps2-8.0b2-linux-x64.flatpak`
3. Open the application after the installation completes.

**Expected behavior**
The user interface should be clear and concise, not filled with broken ASCII characters.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Kubuntu x86_64
 - Version: Linux 6.11.0-13-generic
More information can be found here (thanks to fastfetch)

```txt
OS: Kubuntu x86_64
Host: Modern 15 A11M (REV:1.0)
Kernel: Linux 6.11.0-13-generic
Uptime: 2 hours, 41 mins
Packages: 2600 (dpkg), 205 (nix-user), 48 (nix-default), 9 (flatpak), 12 (snap), 80 (brew), 2 (am), 1 (pacstall)
Shell: fish 3.7.1
Display (ASUS VP228): 1920x1080 @ 60 Hz in 22" [External]
DE: KDE Plasma 6.1.5
WM: KWin (X11)
WM Theme: Breeze
Theme: Breeze (Dark) [Qt], Breeze-Dark [GTK2], Breeze [GTK3/4]
Icons: breeze-dark [Qt], breeze-dark [GTK2/3/4]
Font: Inter Display (10pt) [Qt], Inter Display (10pt) [GTK2/3/4]
Cursor: breeze (24px)
Terminal: konsole 24.8.1
Terminal Font: JetBrainsMono Nerd Font (14pt)
CPU: 11th Gen Intel(R) Core(TM) i7-1195G7 (8) @ 5.00 GHz
GPU: Intel Iris Xe Graphics @ 1.40 GHz [Integrated]
Memory: 7.73 GiB / 15.33 GiB (50%)
Swap: Disabled
Disk (/): 65.81 GiB / 476.64 GiB (14%) - btrfs
Battery (MS-1552): 100% [AC Connected]
Locale: en_CA.UTF-8
```

**Additional context**
Could it be related to **Window Manager** or the **Desktop Environment** I am using?

cyanfish, 2025-01-20:
I'm not seeing the same issue with Plasma 5 on Kubuntu 22.04. I'll try and spin up a 24.10 VM to test at some point.

cyanfish, 2025-01-31:
I'm not able to reproduce with a fresh Kubuntu 24.10 with Plasma 6.1.5. It might be something to do with your font setup?

FlawlessCasual17, 2025-02-01:
Maybe, it's a problem related to my font configuration. Do you know a fix for that?

cyanfish, 2025-02-01:
Something like https://github.com/flatpak/flatpak/issues/5288 maybe

closed 2025-02-11 by FlawlessCasual17