cyanfish/naps2 #532: Support compressed archives (zip, cbz, etc)

source124, 2025-01-21:
**Is your feature request related to a problem? Please describe.**

I have to manually unpack zipped files before importing into NAPS2. And repack them after editing.

**Describe the solution you'd like**

Zipped archives are very convenient to handle scanned material. It would be awesome if NAPS2 supported them.

**Describe alternatives you've considered**

Manually as mentioned.

**Additional context**
Add any other context or screenshots about the feature request here.

cyanfish, 2025-01-26:
I probably won't implement automatic repacking but it shouldn't be too difficult to allow importing zip files.

source124, 2025-01-26:
Thank you. That's half-way there :)

As for packing, you don't have to implement it in the app. Just allow a setting to locate the 7z executable and call it.

cyanfish, 2025-01-26:
https://github.com/cyanfish/naps2/issues/106

cyanfish, 2025-02-02:
Added in [8.0b3](https://www.naps2.com/download).

closed 2025-02-02 by cyanfish