cyanfish/naps2 #535: NAPS CLI: Error: You don't have permission to save files at this location

muescha, 2025-01-22:
**Describe the bug**

I got the error: `You don't have permission to save files at this location`

**To Reproduce**
Steps to reproduce the behavior:
1. add the alias for `naps2.console` to `.zshrc`
2. I done a scan with:

```shell
$ pwd
/Users/muescha/Documents/Personen/Muescha

$ naps2.console -p "7320 Einzug SW 200dpi" -o "test.pdf"
You don't have permission to save files at this location.

$ naps2.console -p "7320 Einzug SW 200dpi" -o "~/Documents/Personen/Muescha/test.pdf"
You don't have permission to save files at this location.
```
3. See error:
`You don't have permission to save files at this location.`

**Expected behavior**
I do not see this error.

**Desktop (please complete the following information):**
 - OS: macOS Sonoma 14.5 (23F79)
 - Version: NAPS 7.5.1.0

**Additional context**
I tried to change the alias to include `sudo`:
```shell
alias naps2.console="sudo /Applications/NAPS2.app/Contents/MacOS/NAPS2 console"
```

But then it scan the sheet but then it stops with no error. I have to cancel it with `control+c`

```

muescha, 2025-01-22:
**Additional context**

### sudo

I tried to change the alias to include `sudo`:
```shell
alias naps2.console="sudo /Applications/NAPS2.app/Contents/MacOS/NAPS2 console"
```

But then it scan the sheet but then it stops with no error. I have to cancel it with `control+c`. After this, the scanner is no longer functional. A second scan also fails and hangs. I have to turn the scanner off and back on to make it work again.

### Autosave with prompt for file path

I also tried to use the `-a` option. I set the autosave to `prompt for file path`. But no luck. It scanned but then no more reactions.

 
```shell
$ alias
naps2.console='/Applications/NAPS2.app/Contents/MacOS/NAPS2 console'

$ naps2.console -p "7320 Einzug SW 200dpi" -a
```

### Autosave with prompt with file pattern

at least setting autosave to use a predefined path works:
```
/Users/muescha/Documents/Personen/Muescha/$(YYYY)-$(MM)-$(DD)-$(hh)-$(mm)-$(ss)-scan.pdf
```

```shell
$ alias
naps2.console='/Applications/NAPS2.app/Contents/MacOS/NAPS2 console'

$ naps2.console -p "7320 Einzug SW 200dpi" -a

ll
Permissions   Size User    Date Modified    Name
.rw-r--r--  15,264 muescha 2025-01-22 18:02 2025-01-22-18-02-53-scan.pdf
```

cyanfish, 2025-01-26:
It looks like the issue is that the working directory is `/Applications/NAPS2.app/` and `~` isn't expanded properly. If you put a full path (e.g. `/Users/muescha/test.pdf`) it should work.

cyanfish, 2025-01-30:
For `~` it seems like zsh doesn't expand it when it's in quotes: https://superuser.com/a/594076

Removing the quotes around the file path should work as long as there are no spaces.

muescha, 2025-02-03:
Thanks for the answer :)

### Using Full Path

```shell
naps2.console -p "7320 Einzug SW 200dpi" -o "/Users/muescha/Documents/Personen/Muescha/test.pdf"
```
I realized I missed the option to avoid using `~` :) This is now working as expected.

### Use current working directory
When specifying `-o "test.pdf"`, the file should be saved in the current working directory.

### Error for `~/` in paths
Using `-o "~/Documents/Personen/Muescha/test.pdf"` should trigger an error message indicating that tilde expansion is unsupported.

### Display accurate path in error messages

It would be more helpful if the error message `You don't have permission to save files at this location.` showed the **actual resolved path** to help users quickly identify mistakes:  

```shell
$ naps2.console -p "7320 Einzug SW 200dpi" -o "test.pdf"
You don't have permission to save files at this location:
/Applications/NAPS2.app/test.pdf

$ naps2.console -p "7320 Einzug SW 200dpi" -o "~/Documents/Personen/Muescha/test.pdf"
You don't have permission to save files at this location:
/Applications/NAPS2.app/~/Documents/Personen/Muescha/test.pdf
```

### Update Documentation at [https://www.naps2.com/doc/command-line](https://www.naps2.com/doc/command-line)

> `-o, --output {path}`  
> Specifies the path, name, and extension of the output file. The extension determines the output type (e.g., ".pdf" for a PDF file, ".jpg" for a JPEG image, etc.).  
> **macOS: Avoid using `~/` in paths. Use the full path, for example, `/Users/username/`.**