cyanfish/naps2 #536: wrong description for --delay

muescha, 2025-01-22:
**Describe the bug**

The description of the `-d, --delay` option is incorrect.

**To Reproduce**

Perform a scan using the `-d 10000` option:
- It waits 10 seconds before starting.
- It waits 10 seconds between scans.

**Expected behavior**
1. Start immediately and only wait between scans.
2. Alternatively, update the description to:  
   `  -d, --delay                (Default: 0) The delay (in milliseconds) before each scan.`
3. maybe add a new option for delay before first scan.

**Desktop (please complete the following information):**
- OS: macOS Sonoma 14.5 (23F79)
- Version: NAPS 7.5.1.0

muescha, 2025-01-22:
you can see here that it wais also 10000ms before each scan:

```shell
naps2.console -p "7320 Einzug SW 200dpi" -a -n 2 -d 10000 -v --progress
2025-01-22 18:50:16.079 NAPS2[19087:1510821] WARNING: Secure coding is automatically enabled for restorable state! However, not on all supported macOS versions of this application. Opt-in to secure coding explicitly by implementing NSApplicationDelegate.applicationSupportsSecureRestorableState:.
Beginning scan...
Waiting 10000ms...
Starting scan 1 of 2...
0 page(s) scanned.
Waiting 10000ms...
Starting scan 2 of 2...
0 page(s) scanned.
```

cyanfish, 2025-01-26:
I think a reasonable solution here would be to add a --first-now option that can be combined with either --delay or --wait-scan to skip the first delay/wait.

cyanfish, 2025-02-02:
Implemented --firstnow in [8.0b3](https://www.naps2.com/download).

closed 2025-02-02 by cyanfish