cyanfish/naps2 #541: CLI: better handling for scanning 2 sided documents

muescha, 2025-01-22:
**Is Your Feature Request Related to a Problem? Please Describe.**

I attempted to scan a double-sided multi-sheet document using a scanner without duplex capability, but the process was more complicated than expected. I had anticipated an experience similar to the GUI, where I could follow simple steps to complete the scan.

The intended workflow was:  
1. Scan the front side (odd pages).  
2. Scan the back side (even pages).  
3. Merge and sort the scans with `--altinterleave`.  
4. Save the output using `-o`.  

However, several issues arose during the process:  
- Using `-n 2` created two separate files instead of one combined document.  
- There was no "wait for keypress" option to pause between scans; the `--wait` flag only delays at the end of the scanning process.  

**Describe the Solution You’d Like**

- Introduce an `--duplex` or `--simulate-duplex` option to streamline the process of scanning double-sided documents manually.  
- Add a `--waitbeetweenscans` option to pause for a keypress during scans, giving users adequate time to prepare the next set of sheets and then resume by pressing Enter.  
- Include an option to combine all scans into a single document by default, allowing for sorting later with `--altinterleave`. Suggested flag names: `--join` or `--join-scans`.

cyanfish, 2025-01-26:
"-n 2" should create a single file unless you use one of the --split options. Was this another case of using --autosave?

cyanfish, 2025-02-02:
Implemented --waitscan in [8.0b3](https://www.naps2.com/download).

closed 2025-02-02 by cyanfish

muescha, 2025-02-03:
> "-n 2" should create a single file unless you use one of the --split options. Was this another case of using --autosave?

yes - the command created 2 separate files for each scan