cyanfish/naps2 #544: CLI: use a buffer

muescha, 2025-01-22:
**Is your feature request related to a problem? Please describe.**

It would be helpful to have a buffer function for scripting purposes.

**Describe the solution you'd like**

Add an option to “save” previous scans in a buffer, collecting all scans into one place. This would allow for manual steps, similar to what is possible in the GUI.

```shell
$ cat scan.sh
naps2.console --buffer --delete-buffer
naps2.console --buffer -p "my profile feeder" --verbose
naps2.console --buffer -p "my profile glass" --verbose
naps2.console --buffer -p "my profile glass" --verbose
naps2.console --buffer -o "~/Documents/Scan/$(YYYY)-$(MM)-original-scan.pdf"
naps2.console --buffer --altinterleave
naps2.console --buffer -o "~/Documents/Scan/$(YYYY)-$(MM)-ordered-scan.pdf"
naps2.console --buffer --close-buffer

$ ./scan.sh

$ naps2.console --buffer --delete-buffer
created buffer
total pages in buffer: 0 page(s)
$ naps2.console --buffer -p "my profile feeder" --verbose
Scan 1 of 1
3 page(s)
saved to buffer
total pages in buffer: 3 page(s)
$ naps2.console --buffer -p "my profile glass" --verbose
Scan 1 of 1
1 page(s)
saved to buffer
total pages in buffer: 4 page(s)
$ naps2.console --buffer -o "~/Documents/Scan/$(YYYY)-$(MM)-original-scan.pdf"
saved 4 page(s) to file as:
~/Documents/Scan/2025-01-original-scan.pdf
$ naps2.console --buffer --import "~/Documents/Scan/last-sheet-with-info.pdf" --verbose
Import file:
  ~/Documents/Scan/last-sheet-with-info.pdf
Imported 2 page(s)
saved to buffer
total pages in buffer: 6 page(s)
$ naps2.console --buffer --altinterleave
reordered 6 page(s) with method: alt interleave
$ naps2.console --buffer -o "~/Documents/Scan/$(YYYY)-$(MM)-final-document.pdf"
saved file as:
~/Documents/Scan/2025-01-final-document.pdf
$ naps2.console --buffer --close-buffer
buffer closed
```

**Describe alternatives you've considered**

I believe this can be achieved in separate steps by using the --import option.

cyanfish, 2025-01-26:
Using a PDF file as a buffer (i.e. `--import a.pdf --output a.pdf`) is this intended way to do this. Granted that's less efficient but I probably won't take on the extra complexity of a separate buffer system.

muescha, 2025-02-03:
>  but I probably won't take on the extra complexity of a separate buffer system.

I agree, it gets complex quickly.