cyanfish/naps2 #550: CLI: open file in NAPS2 GUI (`--open` / `--load`)

muescha, 2025-02-03:
**Is your feature request related to a problem? Please describe.**

I have a working CLI command for a duplex scan:

```shell
naps2.console -v -n 2 --waitscan --altinterleave -p "7320 Einzug SW 200dpi" -o "/Users/muescha/Documents/Personen/Muescha/test.pdf"
Beginning scan...
Press Enter to scan.

Starting scan 1 of 2...
Scanned page 1.
Scanned page 2.
Scanned page 3.
3 page(s) scanned.
Press Enter to scan.

Starting scan 2 of 2...
Scanned page 4.
Scanned page 5.
Scanned page 6.
3 page(s) scanned.
Exporting...
Exporting page 1 of 6.
Exporting page 2 of 6.
Exporting page 3 of 6.
Exporting page 4 of 6.
Exporting page 5 of 6.
Exporting page 6 of 6.
Successfully saved PDF file to /Users/muescha/Documents/Personen/Muescha/test.pdf
```

However, sometimes I end up with blank pages that I want to delete, or I need to perform additional tasks such as rotating pages or manually running OCR.

I expected this to be possible by simply not specifying a post-processing command:

```shell
$ naps2.console -v -n 2 --waitscan --altinterleave -p "7320 Einzug SW 200dpi"
At least one of the -o/--output, -e/--email, or -a/--autosave options must be specified.
```

**Describe the solution you'd like**

A CLI command without any of the `-o/--output`, `-e/--email`, or `-a/--autosave` options should open the GUI.

**Describe alternatives you've considered**

1) Opening a file with a single command

I attempted to open a PDF in NAPS2 via the CLI but couldn’t find a way to do so.

A possible solution could be an option to open a file in the GUI without initiating a scan when using just the `--open` command with a filename (alternatively, `--load` could be used to differentiate it more clearly from `-o/--output`):

```shell
$ naps2.console -v --open "/Users/muescha/Documents/test.pdf"
Loaded file into NAPS: /Users/muescha/Documents/test.pdf
```

2) Opening the file after scanning is complete

If placeholders are used for filenames, it can be difficult to determine the exact output file name when performing a normal CLI scan. To address this, an additional `--open` option without a pathname could be introduced to automatically open the last scanned file in the GUI:

```shell
$ naps2.console -v -n 2 --waitscan --altinterleave -p "7320 Einzug SW 200dpi" -o "scan_{date}.pdf" --open
Opened last scanned file: scan_2024-02-03.pdf
```

cyanfish, 2025-02-03:
`/Applications/NAPS2.app/Contents/MacOS/NAPS2 /file/path`

danielewyss, 2025-02-04:
it would be really nice to have a command line for the NAPS2 app (Not the .console) to open and automatically import a pdf file on the GUI version.

muescha, 2025-02-04:
> `/Applications/NAPS2.app/Contents/MacOS/NAPS2 /file/path`

I see two drawbacks:

1) Full path is required

It doesn't work when I do this:
```shell
$ pwd
/Users/muescha/Documents/Personen/Muescha

$ /Applications/NAPS2.app/Contents/MacOS/NAPS2 test.pdf
```

But this works:
```shell
$ /Applications/NAPS2.app/Contents/MacOS/NAPS2 /Users/muescha/Documents/Personen/Muescha/test.pdf
```

2) New instance issue

Running this command opens a new instance of NAPS2. If NAPS2 is already open, it creates another instance. Pressing `Ctrl+C` in the console then closes the application.

cyanfish, 2025-02-05:
Try `open -a NAPS2 /path`

danielewyss, 2025-02-05:
> But this works:
> 
> $ /Applications/NAPS2.app/Contents/MacOS/NAPS2 /Users/muescha/Documents/Personen/Muescha/test.pdf

I did not expect to make it without an argument, but it works thanks a lot!

cyanfish, 2025-02-26:
I've added doc for this [here](http://localhost:60483/doc/command-line#open-in-gui). Re: setting the output path, there's an existing request to populate the Save PDF dialog based on the last imported file which I plan to implement at some point.

closed 2025-02-26 by cyanfish

muescha, 2025-02-27:
https://www.naps2.com/doc/command-line#open-in-gui

muescha, 2025-02-27:
I tested opening it on my machine.

Note: you should add an info for macOS, that you need to use the full path.

```shell
pwd
/path/to/

# not works
open -a NAPS2 doc.pdf

# works
open -a NAPS2 /path/to/doc.pdf
```

muescha, 2025-03-03:
I can not find the "single instance" option in the settings

[image]

muescha, 2025-03-03:
ok I see it is disabled for macOS?

https://github.com/cyanfish/naps2/blob/94a9309367a26616429e5e395565ff7fae66ffea/NAPS2.Sdk/Platform/MacSystemCompat.cs#L29

cyanfish, 2025-03-03:
Yes, as the doc says it's for Windows/Linux only. On Mac the "open" command should use the existing instance regardless.

muescha, 2025-03-04:
> Yes, as the doc says it's for Windows/Linux only. On Mac the "open" command should use the existing instance regardless.

I should read the doc more carefully.

I tested an on macOS it is a single instance.

PS: maybe this would be better:

> To import a PDF or other file into the NAPS2 GUI you can use these commands. 
>
> Windows/Linux only: Check "Single Instance" in the "Settings" window if you want to open in existing windows instead of opening a new one.