cyanfish/naps2 #551: Searchable PDFs Not Working

winman3000, 2025-02-03:
# Searchable PDFs Not Working

## Description
NAPS2 does not create searchable PDF files, even when the option to create searchable PDFs is selected in the settings.

### Steps to Reproduce
1. Scan a document using NAPS2.
1. Select **"Save PDF"**.
1. Open the saved PDF in Adobe Acrobat.

#### Expected Behavior

The PDF should contain selectable and searchable text.

#### Actual Behavior

The PDF is marked as searchable but only contains an image, making text selection and search impossible.

### Environment
- **NAPS2 Version:** 8.0.3+e7cf25fa120c30decd76c41030dfb65b1ae5c032
- **Operating System:** Windows 11 Professional Version 24H2 (OS Build 26100.3037)
- **OCR Language:** German

### Additional Notes
- The issue persists across different scanned documents.
- The problem occurs regardless of the OCR language selection.

cyanfish, 2025-02-06:
Can you see if there are any [error logs](https://www.naps2.com/doc/troubleshooting#error-log)? And attach a sample PDF with the issue here?

winman3000, 2025-02-07:
Unfortunately there are no error logs, the text is simply exported. I have attached a file that I created with Naps2. The document is marked as “searchable” for screen readers, but there is only one graphic. If you want to export it as text with Adobe Acrobat, no text comes out. It doesn't matter which document you scan.

[TestFile.pdf](https://github.com/user-attachments/files/18702300/TestFile.pdf)

cyanfish, 2025-02-08:
That PDF is searchable for me, works fine with Adobe when I Ctrl+A and Ctrl+C.

winman3000, 2025-02-08:
Strange, but you can't read the PDF with a screen reader. It looks like a graphic. If I go to “Export as text” in Adobe Acrobat, no text is exported either. In other searchable PDF files, the text is exported.

winman3000, 2025-02-09:
I have now tested the problem with three different screen readers: JAWS, NVDA and Narrator, the screen reader from Microsoft that comes with Windows.

The quickest and easiest way to test it is with Narrator. It is important that you follow the steps in this way.

1. Start Narrator with CTRL+Windows+Enter.
1. Open the attached PDF file with Adobe Acrobat.
1. If necessary, confirm the accessibility settings.
1. Now read the document using the arrow keys.

Unfortunately, you cannot read the file as it is a graphic.

cyanfish, 2025-02-11:
Alas, ChatGPT can only make uninformed guesses. But I do have an idea of how it could maybe be fixed.

winman3000, 2025-02-11:
@cyanfish: Thank you!

This is actually the information I received from the JAWS developers and the user community. I only had the text generated by ChatGPT because I have problems formulating English texts.

micceo, 2025-02-22:
Version 8.0b3. Opening the resulting saved PDF in PDF24 viewer no text is delectable. Went back to 7.5.3 and it worked.
Please tell if I can / should provide more info.

7enso, 2025-09-09:
https://github.com/cyanfish/naps2/issues/610

Same issue as the OCR bug which has apparently been fixed?

winman3000, 2026-10-16:
Hi @cyanfish,

After a long period of investigating this issue, I believe I may finally have identified the underlying cause. More importantly, I was able to confirm a possible solution through an actual test with Adobe Acrobat Reader and a screen reader.

### 1. What I discovered

I examined the internal structure of a searchable PDF generated by NAPS2. The OCR text was definitely present, including its Unicode mappings. This explains why the document was searchable and why other PDF viewers, such as Microsoft Edge and Google Chrome, could expose its text to screen readers.

However, Adobe Acrobat Reader continued to expose the document almost exclusively as an image, making it impossible to read properly with JAWS or NVDA.

The interesting part is how NAPS2 embeds the OCR text.

As far as I can tell from the source code, the PDFsharp-based export currently uses `XGraphics.DrawString()` with `XBrushes.Transparent` to create the invisible OCR text layer.

This is different from using the PDF specification's dedicated invisible text rendering mode.

### 2. The solution I successfully tested

I modified the existing PDF without performing OCR again.

Instead of relying on transparent or white text, I changed the OCR text rendering to use the PDF operator:

`3 Tr`

This activates text rendering mode 3, which means that the text is neither filled nor stroked. It remains invisible while retaining its underlying text information.

I deliberately preserved:

- The original OCR text, including all recognition errors.
- The existing text positions.
- The original scanned image.
- The existing font and Unicode mappings.

I did not add PDF tags or perform any new text recognition.

**The result: Adobe Acrobat Reader was finally able to expose the OCR text to my screen reader!**

The actual text was still largely incorrect because the original scan had been recognized upside down, but that is a completely separate OCR-quality issue.

The important finding is that the previously inaccessible text became readable in Acrobat simply by changing how the existing OCR text was rendered.

This strongly suggests that the issue is related to the PDF text rendering method rather than the absence of PDF tags.

Of course, this is a successful test with one document rather than definitive proof that the same change will resolve every affected PDF.

### 3. Possible short-term fix

Would it be possible to modify the PDFsharp-based export so that OCR text is written using the PDF text rendering mode `3 Tr`, rather than relying on `XBrushes.Transparent`?

I also noticed that the PDFium-based export already uses `TextRenderMode.Invisible`, so perhaps the PDFsharp implementation could be adjusted to produce equivalent PDF content.

Ideally, this would require only a relatively small modification to the PDF generation code, without replacing the OCR engine or introducing a complete PDF tagging implementation.

Given the results of my test, I believe this would be worth investigating.

I would be happy to provide both the original PDF and the modified, screen-reader-accessible version for comparison.

### 4. Long-term improvement: Tagged PDFs and PDF/UA accessibility

Beyond this immediate issue, I would also like to suggest a longer-term accessibility improvement.

It would be wonderful if NAPS2 could eventually generate properly tagged PDFs, ideally following the PDF/UA standard.

While a searchable text layer is an important first step, it does not provide the semantic structure that screen reader users need for efficient document navigation.

For example, a properly tagged PDF could allow screen readers to:

- Identify and navigate headings at different levels.
- Recognize paragraphs and their logical reading order.
- Identify numbered and bulleted lists.
- Recognize tables, including rows, columns and header cells.
- Distinguish meaningful content from decorative images.
- Navigate document structures instead of reading everything as unstructured text.

I understand that this is significantly more complex than simply embedding an OCR text layer.

It also depends on what structural information Tesseract can actually provide.

Tesseract already supports hOCR output, which contains information about recognized text regions, paragraphs, lines, words and their positions. Some of this information might be useful as a starting point for generating a logical PDF structure.

However, I realize that recognizing actual heading levels, lists and especially complex tables requires much more than ordinary OCR. Tesseract's layout information alone may not be sufficient to reliably identify these semantic elements.

Perhaps a first step could be generating tagged PDFs with a correct reading order, paragraphs and appropriate document language information, while adding more advanced structure recognition later.

The scanned page image could also be marked as a PDF artifact when its textual content is already represented by the accessible OCR layer, avoiding unnecessary image announcements by screen readers.

I also noticed that newer versions of PDFsharp provide accessibility-related functionality, although I understand that NAPS2 currently uses a customized PDFsharp implementation and that migrating to a newer version may not be straightforward.

### 5. Why this matters

As someone who is completely blind and relies on screen readers every day, I consider this a particularly important issue.

NAPS2 is an excellent scanning application, and I would love to see it become a reliable solution for creating documents that are not only searchable but also genuinely accessible.

The good news is that the immediate Acrobat Reader compatibility problem may be much easier to solve than we originally thought.

Thank you again for your work on NAPS2 and for taking the time to investigate this issue.

I hope these findings are helpful!

cyanfish, 2026-10-16:
@winman3000 Can you try with NAPS2 8.5.0? This was already updated to use Tr etc. (for other reasons, but might fix this as well)

winman3000, 2026-10-16:
@cyanfish: I tested NAPS2 8.5.1 as you suggested, and I'm happy to confirm that **the original issue has been resolved!** PDFs generated by NAPS2 are now readable with JAWS and NVDA in Adobe Acrobat Reader. Thank you so much for fixing this!

There is just one thing I noticed that could perhaps be improved in the future: the PDFs still lack accessibility tags and a proper document structure.

For example, I tested a scanned form containing two separate address blocks positioned next to each other. Although the OCR recognized most of the text correctly, the reading order was somewhat confusing, with one address block appearing after unrelated content.

I wonder if NAPS2 could make use of Tesseract's hOCR output to preserve text blocks, paragraphs and their logical reading order. In the longer term, it would be fantastic if NAPS2 could also generate tagged PDFs, allowing screen readers to recognize headings, lists and tables, provided Tesseract or additional layout analysis can identify these elements reliably.

I realize this would be a separate and more complex feature, but even basic PDF tagging would make a big difference for accessibility.

Most importantly, **the original problem is fixed**, and I'm really grateful for your work. As a blind user who relies on screen readers every day, I truly appreciate it!

Thank you again!

cyanfish, 2026-10-25:
Glad that's working, opened https://github.com/cyanfish/naps2/issues/941 to track the other request.

closed 2026-10-25 by cyanfish