cyanfish/naps2 #552: Better OCR Overlay

pschopen, 2025-02-03:
NAPS is one of the fastest and most convenient tools I know for optimising and OCRing PDFs. However, often the OCRed text doesn't match the text in the underlying image.

This, on the other hand, works quite well with this tool: https://github.com/UB-Mannheim/zotero-ocr Maybe it makes sense to use the same approach? I don't know the technical details, it's just an idea.

cyanfish, 2025-03-29:
Do you have an example PDF where the text doesn't match up?

pschopen, 2025-04-03:
It seems to happen when the document uses different fonts. In the following example, the text is correct, but the heading is not.

[OCR Overlay.pdf](https://github.com/user-attachments/files/19582430/OCR.Overlay.pdf)

cyanfish, 2026-10-12:
Please try with [8.5.0](https://www.naps2.com/download) as I think it should be fixed. Let me know if there are any docs still with issues (and please upload a sample if so).

closed 2026-10-12 by cyanfish