cyanfish/naps2 #553: OCR for JPG

scottf007, 2025-02-05:
Hello, 

I have been using your application for about 5 years now, and love it. 

I have recently migrated to the windows store app - it seems to work really well. 

I scan my letters, finance stuff etc -  and store it as a JPG unless there are a lot of pages then I convert it to PDF, the reason I do this is that JPG seems to be more of a base level format instead of a jpg wrapped in a PDF. Easier to open and scroll through 20 documents.

I would love if you could extend the OCR to JPG, you could just dump the whole lot into EXIF of the image, and windows picks this up (I can test debian and ubuntu if you wanted to move forward).

Thanks for your hard work, you have made a great product. 

Cheers
Scott

cyanfish, 2025-04-19:
Is there a standard for including OCR'd text in EXIF? Something I'm more likely to do is provide a way to export a text file that you could save alongside your jpeg.

IvarSnaaijer, 2026-10-20:
adding a sidecar with text from the image is not really the problem. it can be implemented automatically, if you ask for an image, currently the ocr is ignored. I can make it so it will just generate the txt file if you ask. but there are also Global OCR settings. these are currently only applicable to pdf's i do have to modify the ExportController.cs  to make sure the behavior does not change.
It will still do if someone left the parameter unintended. nice surprise after an upgrade..

IvarSnaaijer, 2026-10-20:
Added tests to make sure the behavior does not change when it is not intended. Again a bit bigger, but i think it should not suddenly start generating txt files, but now the user must explicitly select the option.