cyanfish/naps2 #556: Won't run on MacOS Sequoia 15.3

ttfna, 2025-02-11:
Hi

I have tried installing v7.5.3 and 8.0b3 (universal and apple silicone) on my M4 MacBook Pro running Sequoia 15.3 but without success.  The installation completes but on running the app nothing happens (it looks like it may run and quit)

If I show the package contents and then run MacOS/NAPS2 I get the following output:

Last login: Tue Feb 11 17:16:15 on ttys000
/Applications/NAPS2.app/Contents/MacOS/NAPS2 ; exit;                            
****@****~ % /Applications/NAPS2.app/Contents/MacOS/NAPS2 ; exit;
Unhandled exception. System.UnauthorizedAccessException: UnauthorizedAccess_IODenied_Path, /Users/***/.config/naps2
 ---> System.IO.IOException: Permission denied
   Exception_EndOfInnerExceptionStack
   at System.IO.FileSystem.CreateDirectory(String, UnixFileMode)
   at System.IO.FileSystem.CreateDirectory(String)
   at System.IO.Directory.CreateDirectory(String)
   at NAPS2.Paths.EnsureFolderExists(String folderPath)
   at NAPS2.Paths.get_AppData()
   at NAPS2.Program.Main(String[] args)
zsh: abort      /Applications/NAPS2.app/Contents/MacOS/NAPS2

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]

with thanks for any assistance

ttfna, 2025-02-12:
I have just installed this sucessfully on an M1 MacMini also running MacOS Sequoia 15.3, so I wonder if the difference relates to the MacBook Pro being an M4?

with thanks

ttfna, 2025-02-14:
I have just discovered that the same installations will launch from the terminal using:

sudo /Applications/NAPS2.app/Contents/MacOS/NAPS2

It won't work without sudo

cyanfish, 2025-02-14:
It might work to grant full disk access https://www.google.com/search?q=macos%20grant%20full%20disk%20access

Though I'm not sure why it's needed for this particular device

ttfna, 2025-02-14:
I've just thought to run it on a different user account on the same machine and it runs without a problem.  So there must be something odd/corrupted with the original user account.

Sorry to raise a concern that is not down to you!

closed 2025-02-14 by ttfna