cyanfish/naps2 #557: First time testing, multiple bugs + crash

theRolento, 2025-02-13:
**Describe the bug**
1. Save PDF just crash the program.
2. No errorlog is generated.
3. No debuglog is generated (option checked)

**To Reproduce**
Steps to reproduce the behavior:
1. Import PDF (410 pages)
2. Enable OCR settings (tried every combination with languages and mode)
3. Save PDF
4. Stuck on 0 for some time and it just crash
5. Check Data folder (portable) no errorlog and no debuglog

**Expected behavior**
It should generate and save the PDF and it should at least generate the debuglog.

**Desktop (please complete the following information):**
OS: Windows 11
Version: 8.0b3 Portable

theRolento, 2025-02-14:
> You can check the [Windows event viewer](https://www.howtogeek.com/123646/htg-explains-what-the-windows-event-viewer-is-and-how-you-can-use-it/) under Application to see if it has any more info.

Found this event:
`		<EventData>
			<Data>Application: NAPS2.exe
CoreCLR Version: 9.0.24.52809
.NET Version: 9.0.0
Description: The process was terminated due to an unhandled exception.
Stack:
   at NAPS2.Pdf.PdfExporter.DrawOcrTextOnPdfiumPage(PdfSharpCore.Pdf.PdfPage, NAPS2.Pdf.Pdfium.PdfDocument, NAPS2.Pdf.Pdfium.PdfPage, NAPS2.Pdf.PdfiumFontSubsets, NAPS2.Ocr.OcrResult)
   at NAPS2.Pdf.PdfExporter.MergePassthroughPages(System.IO.MemoryStream, NAPS2.ImportExport.OutputPathOrStream, System.Collections.Generic.List`1&lt;PageExportState&gt;, NAPS2.Pdf.PdfExportParams, NAPS2.Util.ProgressHandler)
   at NAPS2.Pdf.PdfExporter+&lt;&gt;c__DisplayClass8_0+&lt;&lt;Export&gt;b__0&gt;d.MoveNext()
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1+AsyncStateMachineBox`1[[System.Boolean, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[NAPS2.Pdf.PdfExporter+&lt;&gt;c__DisplayClass8_0+&lt;&lt;Export&gt;b__0&gt;d, NAPS2.Sdk, Version=8.0.3.0, Culture=neutral, PublicKeyToken=null]].ExecutionContextCallback(System.Object)
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1+AsyncStateMachineBox`1[[System.Boolean, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[NAPS2.Pdf.PdfExporter+&lt;&gt;c__DisplayClass8_0+&lt;&lt;Export&gt;b__0&gt;d, NAPS2.Sdk, Version=8.0.3.0, Culture=neutral, PublicKeyToken=null]].MoveNext(System.Threading.Thread)
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1+AsyncStateMachineBox`1[[System.Boolean, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[NAPS2.Pdf.PdfExporter+&lt;&gt;c__DisplayClass8_0+&lt;&lt;Export&gt;b__0&gt;d, NAPS2.Sdk, Version=8.0.3.0, Culture=neutral, PublicKeyToken=null]].MoveNext()
   at System.Threading.Tasks.AwaitTaskContinuation.RunOrScheduleAction(System.Runtime.CompilerServices.IAsyncStateMachineBox, Boolean)
   at System.Threading.Tasks.Task.RunContinuations(System.Object)
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1[[System.__Canon, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].SetResult(System.__Canon)
   at NAPS2.Threading.Pipeline+PipelineStep`1+&lt;Run&gt;d__10[[System.__Canon, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].MoveNext()
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1+AsyncStateMachineBox`1[[System.__Canon, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[NAPS2.Threading.Pipeline+PipelineStep`1+&lt;Run&gt;d__10[[System.__Canon, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]], NAPS2.Sdk, Version=8.0.3.0, Culture=neutral, PublicKeyToken=null]].ExecutionContextCallback(System.Object)
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1+AsyncStateMachineBox`1[[System.__Canon, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[NAPS2.Threading.Pipeline+PipelineStep`1+&lt;Run&gt;d__10[[System.__Canon, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]], NAPS2.Sdk, Version=8.0.3.0, Culture=neutral, PublicKeyToken=null]].MoveNext(System.Threading.Thread)
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1+AsyncStateMachineBox`1[[System.__Canon, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[NAPS2.Threading.Pipeline+PipelineStep`1+&lt;Run&gt;d__10[[System.__Canon, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]], NAPS2.Sdk, Version=8.0.3.0, Culture=neutral, PublicKeyToken=null]].MoveNext()
   at System.Threading.Tasks.AwaitTaskContinuation.RunOrScheduleAction(System.Runtime.CompilerServices.IAsyncStateMachineBox, Boolean)
   at System.Threading.Tasks.Task.RunContinuations(System.Object)
   at System.Threading.Tasks.Task`1[[System.Threading.Tasks.Dataflow.Internal.VoidResult, System.Threading.Tasks.Dataflow, Version=9.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a]].TrySetResult(System.Threading.Tasks.Dataflow.Internal.VoidResult)
   at System.Threading.Tasks.TaskCompletionSource`1[[System.Threading.Tasks.Dataflow.Internal.VoidResult, System.Threading.Tasks.Dataflow, Version=9.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a]].TrySetResult(System.Threading.Tasks.Dataflow.Internal.VoidResult)
   at System.Threading.Tasks.Dataflow.Internal.TargetCore`1[[System.__Canon, System.Private.CoreLib, Version=9.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].CompleteBlockOncePossible()
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(System.Threading.Thread, System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(System.Threading.Tasks.Task ByRef, System.Threading.Thread)
[... cut ...] lockOncePossible()&#10;   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(System.Threading.Thread, System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)&#10;   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(System.Threading.Tasks.Task ByRef, System.Threading.Thread)&#10;   at System.Threading.ThreadPoolWorkQueue.Dispatch()&#10;   at System.Threading.PortableThreadPool+WorkerThread.WorkerThreadStart()&#10;</Message>
			<Level>Errore</Level>`

theRolento, 2025-02-14:
Same thing happens with 7.5.3.
Basically i can't use naps2

cyanfish, 2025-02-23:
Can you try this [test version](https://1drv.ms/u/c/2918c6d8dc4b6479/Eac9eOwcLi5Gpy6QzPdHf2cBEmHpVM7TaiMoaIrP-ZpwJA?e=OfHo0E) and see if it makes any difference?

theRolento, 2025-02-23:
> Can you try this [test version](https://1drv.ms/u/c/2918c6d8dc4b6479/Eac9eOwcLi5Gpy6QzPdHf2cBEmHpVM7TaiMoaIrP-ZpwJA?e=OfHo0E) and see if it makes any difference?

Nope, just tried. Not working it crashes like before.
If i manually select a bunch of pages apparently works, like 60-70 pages at once. Starting from 80 it's random, it can crash or it can work. Around 100+ pages it always crash.

cyanfish, 2026-10-25:
Several PDF crashes have been fixed as of 8.6.0. (See #918)

closed 2026-10-25 by cyanfish