cyanfish/naps2 #558: OCR CLI will not create searchable PDFs

ttfna, 2025-02-14:
**Describe the bug**

I'm trying to use the command line to make a pdf searchable using OCR .  The command fails with the  error "No device was specified. Either use "--profile" to specify a profile with a device, or use "--device" to choose a particular device (and "--listdevices" to see available choices)."

I have no scanner set up with my system but the equivalent function works using the GUI with no attached scanner.

The bug is present using both NAPS 7.5.3 and 8.0b3

**To Reproduce**
Steps to reproduce the behavior:

At the command line enter the command:
naps2.console -I /PathtoPDF/infile.pdf -o /PathtoPDF/outfile.pdf  

I have tried various options such as including --enableocr and --ocrlang "eng"

**Expected behavior**
For the command to run without needing a scanner profile

**Desktop (please complete the following information):**
MacOS Sequoia 15.3 M1 MacMini

with thanks

cyanfish, 2025-02-14:
See https://www.naps2.com/doc/command-line#combining-and-converting-files

ttfna, 2025-02-14:
Thanks for your prompt reply.  I had overlooked that "-n 0" applied to the OCR option.  It is working now.

with thanks

closed 2025-02-14 by ttfna