cyanfish/naps2 #567: Installer for Windows ARM

rafalkraz, 2025-02-28:
There is no installer for Windows ARM. It would be great if NAPS worked natively on Snapdragon X CPU.

Please compile app for ARM Windows version. Now it runs using emulation.

[image]

fabalorian, 2025-03-29:
+1
That would be great for all Suface Pro 11 users

martinsuchan, 2025-04-14:
I have Lenovo Yoga Slim 7x Copilot+ PC with Snapdragon X Elite CPU. I'd appreciate native ARM64 build as well.

Note right now the installer fails to launch on Windows 11 ARM64, I need to use the portable version in emualted mode.
[image]

Also with the app built on .NET Framework, .NET Framework 4.8.1 and current .NET 8.0+ fully supports ARM64 target.

cyanfish, 2025-06-08:
I've added an arm64 installer in NAPS2 [8.2.0](https://www.naps2.com/download). For now only an .exe installer, if you need a .msi or .zip let me know and I'll consider building it too.

closed 2025-06-08 by cyanfish