cyanfish/naps2 #570: Add an option "Open containing folder" in the context menu for the scanned files

dobri-popov, 2025-03-10:
Description:
    After scanning files, users often need to access their location for further actions such as reviewing, copying, or moving files. This would allow users to quickly navigate to the file location without needing to manually search for it.

Suggested Implementation:
    Add a right-click context menu option that opens the containing folder in the system’s file explorer.

cyanfish, 2025-03-12:
When the "PDF Saved" notification appears in the bottom right you can right-click the link and select "Open Folder".

dobri-popov, 2025-03-18:
After scanning a new page, a copy appears on the app's main screen, but its storage location is unclear. Double clicking on the image opens a Preview window which does not give you an option to find where the file is stored.

[image]

cyanfish, 2025-03-20:
This is what I mean, after it's saved. Before you save the file there's nowhere to open. (Well, technically there's a temp file, but that's not going to work reliably.)

[image]

cyanfish, 2025-03-20:
Gotcha. It works with "Save Images" too, but I see your point about making it easier. #70 includes the possibility of an "open in external editor" feature.

cyanfish, 2025-04-19:
Closing in favor of #70 to track "open in external editor".

closed 2025-04-19 by cyanfish