cyanfish/naps2 #574: System.OverflowException: Overflow error when scanning a document with high resolution using ESCL driver

OlegAndreych, 2025-03-17:
Got an exception when scanning a doc with 2400 dpi resolution using ESCL driver.

Scanning with lower resolution (e.g. 1200 works OK).

Excpetion:  
```
2025-03-17 19:23:03.4383 21020 Overflow error. System.OverflowException: Overflow error.
   at void System.Drawing.Graphics.CheckErrorStatus(int status)
   at void System.Drawing.Graphics.DrawImage(Image image, Rectangle destRect, int srcX, int srcY, int srcWidth, int srcHeight, GraphicsUnit srcUnit, ImageAttributes imageAttrs, DrawImageAbort callback, IntPtr callbackData) x 2
   at GdiImage NAPS2.Images.Gdi.GdiImageTransformer.PerformTransform(GdiImage image, ResizeTransform transform)
   at TImage NAPS2.Images.Transforms.AbstractImageTransformer<TImage>.PerformTransform(TImage image, ThumbnailTransform transform)
   at TImage NAPS2.Images.Transforms.AbstractImageTransformer<TImage>.Apply(TImage image, Transform transform)
   at void NAPS2.Scan.Internal.RemotePostProcessor.DoRevertibleTransforms(ref ProcessedImage processedImage, ref IMemoryImage image, ScanOptions options, PostProcessingContext postProcessingContext)
   at ProcessedImage NAPS2.Scan.Internal.RemotePostProcessor.PostProcess(IMemoryImage image, ScanOptions options, PostProcessingContext postProcessingContext)
   at Task NAPS2.Scan.Internal.RemoteScanController.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, PostProcessingContext> callback)+(IMemoryImage image) => { }
   at async Task NAPS2.Scan.Internal.Escl.EsclScanDriver.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<IMemoryImage> callback) x 2
   at async Task NAPS2.Scan.Internal.RemoteScanController.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, PostProcessingContext> callback)
   at async Task NAPS2.Scan.ScanController+<>c__DisplayClass15_1.<Scan>g__DoScan|6(?)+DoScan(?)
   at async IAsyncEnumerable<ProcessedImage> NAPS2.Scan.ScanController.Scan(ScanOptions options, CancellationToken cancelToken)+(?) => { }

```

**To Reproduce**
Steps to reproduce the behavior:
1. Set up an AirScan scanner;
2. Scan A4 with resolution 2400 dpi;
3. See error;

**Expected behavior**
Doc should be scanned.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 7.5.3.0

cyanfish, 2025-03-21:
NAPS2 can currently only handle images that are about 2GB uncompressed, which that slightly exceeds. If you don't need it to be exactly 2400 dpi, in [NAPS2 8.1.0](https://www.naps2.com/download) you can specify a custom dpi and choose something like 2200 which should work.

OlegAndreych, 2025-03-21:
It worked.
Thanks for the nice addition!