cyanfish/naps2 #576: DPI Setting in profiles not retained after upgrade

SelfMan, 2025-03-21:
I have upgraded from v7.5.3 to 8.1.0.
I have several scan profiles, all set to 150Dpi. When I checked the PROFILES configuration, all items were reset to 100Dpi.
I've edited them back to 150Dpi, but they always reset to 100Dpi

When I change the DPI setting within a profile, I can see the change "replcate" in the new side bar. But when I reopen the profile again, it's reset to 100DPI while the side bar still shows the new value.
When I close NAPS2 and reopen it, the scan profile is reset back to 100DPI even in the side bar.
It does not matter if I change the profile's config from the side bar or the PROFILES menu.
It's always reset to 100DPI

- open scan profile which shows 100dpi
- change the value to 150 (or any other)
- see value propagate in the side bar
- re-open the profile's config view - it got reset to 100DPI whi the side bar still shows 150DPI (or other selected)

Windows 10 22H2 Fully up to date
NAPIS2 8.10.0
Multiple MFP Printers used

[image]

SelfMan, 2025-03-21:
I've tried o create new profile, but the behavior is the same. When the "Profile settings" window opens, the DPI selection is reset to 100DPI. Closing and reopening NAPS2 I see the side bar being reset to 100DPI
I've also tried to change the DPI value in the side bar, close NAPS2, reopen it and it was reset again.

cyanfish, 2025-03-21:
Thanks, this should be fixed in 8.1.1.

closed 2025-03-21 by cyanfish