cyanfish/naps2 #577: NAPS2 hangs afterf scanning

akuropka, 2025-03-21:
**Describe the bug**
After scanning, TWAIN dialog does not disappear and the picture is imported several times. NAPS2 hangs.

**To Reproduce**
Steps to reproduce the behavior:
1. Scan a page

**Expected behavior**
TWAIN dialog closes and one picture is imported, NAPS2 is responsive.

**Desktop (please complete the following information):**
 - OS: Microsoft Windows [Version 10.0.26120.3576]
 - Version: 8.1.0

**Additional context**
Does not happen with v7.5.3.
It was already reported in the beta feedback thread: https://github.com/cyanfish/naps2/discussions/467#discussioncomment-12252672

**Screenshots**
https://github.com/user-attachments/assets/a6e0b4ea-b6b3-4abb-936c-775e282a3191

cyanfish, 2025-03-21:
Can you follow [these instructions](https://www.naps2.com/doc/troubleshooting#debug-log) to collect debug logs?

akuropka, 2025-03-21:
[debuglog.txt](https://github.com/user-attachments/files/19394170/debuglog.txt)

cyanfish, 2025-03-21:
Does it work differently if it's flatbed vs feeder? How about if you select "Predefined settings" vs "Native UI" in your profile?

akuropka, 2025-03-22:
Flatbed vs feeder makes no difference.
With native UI NAPS2 hangs, too.

[debuglog.txt](https://github.com/user-attachments/files/19401991/debuglog.txt)

cyanfish, 2025-03-29:
Can you try this [test version](https://1drv.ms/u/c/2918c6d8dc4b6479/EXb7wx7MjfZMnFY-TTexxRIB4qJdPaZYd6yNw7c2knS_5Q?e=tTjC09) and see if it makes any difference?

akuropka, 2025-03-29:
Thank you, @cyanfish, but sadly no success.

[debuglog.txt](https://github.com/user-attachments/files/19522940/debuglog.txt)

acorp77, 2025-04-02:
I've got the same problem with Kyocera Twain driver 3.2.0512, Naps 8.1.2.0. Native mode does nothing.

cyanfish, 2025-04-03:
This should be fixed in [8.1.3](https://www.naps2.com/download).

closed 2025-04-03 by cyanfish

akuropka, 2025-04-03:
It works 👍 thank you 😊

funkerwolf, 2025-04-21:
with kyocera fs-1025mfp still not working

akuropka, 2025-04-21:
I am using Twain over network and it does work now.

cyanfish, 2025-04-24:
@funkerwolf In your profile settings, under Advanced, Twain Implementation, if you select "Native Transfer" does it make a difference?

funkerwolf, 2025-04-24:
No, also error

Elias-Graf, 2025-08-25:
I'm having the same or similar issue with my brother. I'm using the SANE API with the implementation from brscan5. It successfully starts scanning, the printer itself seems to finish without any issue. Naps2 although, seems to be stuck at "Scanning page 1".

Gnomes "simple-scan" does not have this issue, and successful scans alls the documents in the feeder via the brscan5 SANE implementation.

<details>
<summary>Debug log</summary>

```
2025-08-25 23:36:45.2488 6758 Scanning with ScanDevice { Driver = Sane, ID = brother5:net1;dev0, Name = home (brother5), IconUri =  }
2025-08-25 23:36:45.2488 6758 Scan source: Duplex; bit depth: Color; dpi: 300; page size: 210x297 mm
2025-08-25 23:36:45.3227 7330 Opening SANE Device "brother5:net1;dev0"
2025-08-25 23:36:49.5660 7330 Option: 2 mode Scan mode Select the scan mode String None 30 SoftSelect, SoftDetect StringList 24bit Color[Fast],Black & White,True Gray,Gray[Error Diffusion]
2025-08-25 23:36:49.5784 7330 Option: 3 resolution Scan resolution Sets the resolution of the scanned image. Int Dpi 4 SoftSelect, SoftDetect WordList 100,150,200,300,400,600,1200,2400,4800,9600
2025-08-25 23:36:49.5784 7330 Option: 4 source Scan source Selects the scan source (such as a document-feeder). String None 64 SoftSelect, SoftDetect StringList FlatBed,Automatic Document Feeder(left aligned),Automatic Document Feeder(center aligned),Automatic Document Feeder(left aligned,Duplex),Automatic Document Feeder(center aligned,Duplex)
2025-08-25 23:36:49.5784 7330 Option: 5 brightness Brightness Controls the brightness of the acquired image. Fixed Percent 4 SoftSelect, SoftDetect, Inactive Range -50-50/1
2025-08-25 23:36:49.5784 7330 Option: 6 contrast Contrast Controls the contrast of the acquired image. Fixed Percent 4 SoftSelect, SoftDetect, Inactive Range -50-50/1
2025-08-25 23:36:49.5784 7330 Option: 7 MultifeedDetection Multifeed Detection  Bool None 4 SoftSelect, SoftDetect, Inactive, Advanced None
2025-08-25 23:36:49.5784 7330 Option: 8 AutoDocumentSize Auto Document Size  Bool None 4 SoftSelect, SoftDetect, Advanced None
2025-08-25 23:36:49.5784 7330 Option: 9 AutoDeskew Auto Deskew  Bool None 4 SoftSelect, SoftDetect, Advanced None
2025-08-25 23:36:49.5784 7330 Option: 10 SkipBlankPage Skip Blank Page  Bool None 4 SoftSelect, SoftDetect, Advanced None
2025-08-25 23:36:49.5784 7330 Option: 11 SkipBlankPageSensitivity Skip Blank Page sensitivity  Fixed Percent 4 SoftSelect, SoftDetect, Advanced Range 0-100/1
2025-08-25 23:36:49.5791 7330 Option: 13 tl-x Top-left x Top-left x position of scan area. Fixed Mm 4 SoftSelect, SoftDetect Range 0-215.90000762951095/0.09999237048905166
2025-08-25 23:36:49.5791 7330 Option: 14 tl-y Top-left y Top-left y position of scan area. Fixed Mm 4 SoftSelect, SoftDetect Range 0-355.6/0.09999237048905166
2025-08-25 23:36:49.5791 7330 Option: 15 br-x Bottom-right x Bottom-right x position of scan area. Fixed Mm 4 SoftSelect, SoftDetect Range 0-215.90000762951095/0.09999237048905166
2025-08-25 23:36:49.5791 7330 Option: 16 br-y Bottom-right y Bottom-right y position of scan area. Fixed Mm 4 SoftSelect, SoftDetect Range 0-355.6/0.09999237048905166
2025-08-25 23:36:49.5811 7330 Maybe setting source
2025-08-25 23:36:49.5811 7330 Setting source to Automatic Document Feeder(left aligned,Duplex)
2025-08-25 23:36:49.5811 7330 Maybe setting adf_mode
2025-08-25 23:36:49.5811 7330 Maybe setting adf-mode
2025-08-25 23:36:49.5821 7330 Maybe setting mode
2025-08-25 23:36:49.5821 7330 Setting mode to 24bit Color[Fast]
2025-08-25 23:36:49.5848 7330 Maybe setting resolution
2025-08-25 23:36:49.5848 7330 Setting resolution to 300
2025-08-25 23:36:49.5869 7330 Maybe setting page-width
2025-08-25 23:36:49.5869 7330 Maybe setting page-height
2025-08-25 23:36:49.5871 7330 Maybe setting tl-x
2025-08-25 23:36:49.5871 7330 Setting tl-x to 0
2025-08-25 23:36:49.5871 7330 Maybe setting tl-y
2025-08-25 23:36:49.5871 7330 Setting tl-y to 0
2025-08-25 23:36:49.5871 7330 Maybe setting br-x
2025-08-25 23:36:49.5871 7330 Setting br-x to 210
2025-08-25 23:36:49.5871 7330 Maybe setting br-y
2025-08-25 23:36:49.5871 7330 Setting br-y to 297
```
</details>