cyanfish/naps2 #580: OCR no longer works in the portable version 8.1.1

TheBastelWastl, 2025-03-22:
OCR no longer works for me in the portable version of NAPS2 8.1.1. In the portable version 7.5.3.0 this still worked. I am using Windows 10 64bit.

Here is the error message from the errorlog.txt:

2025-03-22 13:19:47.2922 3472 Error running OCR System.IO.FileNotFoundException: Could not find file 'D:\NAPS\Data\temp\2l3qnth3.4gd.hocr'.
File name: 'D:\NAPS\Data\temp\2l3qnth3.4gd.hocr'
   at SafeFileHandle Microsoft.Win32.SafeHandles.SafeFileHandle.CreateFile(string fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options)
   at SafeFileHandle Microsoft.Win32.SafeHandles.SafeFileHandle.Open(string fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options, long preallocationSize, UnixFileMode? unixCreateMode)
   at new System.IO.Strategies.OSFileStreamStrategy(string path, FileMode mode, FileAccess access, FileShare share, FileOptions options, long preallocationSize, UnixFileMode? unixCreateMode)
   at FileStreamStrategy System.IO.Strategies.FileStreamHelpers.ChooseStrategyCore(string path, FileMode mode, FileAccess access, FileShare share, FileOptions options, long preallocationSize, UnixFileMode? unixCreateMode)
   at new System.IO.FileStream(string path, FileMode mode, FileAccess access, FileShare share, int bufferSize)
   at Stream System.Xml.XmlDownloadManager.GetStream(Uri uri, ICredentials credentials, IWebProxy proxy)
   at void System.Xml.XmlTextReaderImpl.FinishInitUriString()
   at XmlReader System.Xml.XmlReaderSettings.CreateReader(string inputUri, XmlParserContext inputContext)
   at XDocument System.Xml.Linq.XDocument.Load(string uri, LoadOptions options)
   at async Task<OcrResult> NAPS2.Ocr.TesseractOcrEngine.ProcessImage(ScanningContext scanningContext, string imagePath, OcrParams ocrParams, CancellationToken cancelToken)

If I watch the D:\NAPS\Data\temp\ folder I see that it creates a .jpg-file after a scan but no .hocr-file.

cyanfish, 2025-03-22:
Are there any other messages in the error log? If you open a terminal in the NAPS2 folder and run `.\App\lib\_win64\tesseract.exe --version` what does it show? Do you have a sample PDF that OCRs correctly in 7.5.3 but not 8.1.1?

TheBastelWastl, 2025-03-23:
Hi, here the version message of tesseract.exe from version 8.1.1.0:

D:\NAPS>.\App\lib\_win64\tesseract.exe --version
tesseract 5.5.0
 leptonica-1.85.0 (Jan 18 2025, 15:08:22) [MSC v.1940 LIB Release x64]
  libgif 5.2.2 : libjpeg 9f : libpng 1.6.45 : libtiff 4.7.0 : zlib 1.3.1 : libwebp 1.5.0 : libopenjp2 2.5.3
 Found AVX2
 Found AVX
 Found FMA
 Found SSE4.1

In 7.5.3, every OCR scan works perfectly, in 8.1.1 I always get the same error message (see attached screenshot. Version 7.5.3 creates two files during the scan in the NAPS\data\temp folder  (.png and .hocr, Version 8.1.1 creates only on file (.png). I also attached two sample PDF-files.

[image]
[Scan_OCR_v7.5.3.0.pdf](https://github.com/user-attachments/files/19408506/Scan_OCR_v7.5.3.0.pdf)
[Scan_OCR_v8.1.1.0.pdf](https://github.com/user-attachments/files/19408505/Scan_OCR_v8.1.1.0.pdf)

cyanfish, 2025-03-23:
Can you try this [test version](https://1drv.ms/u/c/2918c6d8dc4b6479/EXQ90JwRum1NgHsPHkE_-hMBV77hsITpnLBdHYnq2i2b0A?e=Io9WGX), try running OCR, then attach the errorlog.txt here?

TheBastelWastl, 2025-03-23:
Thanks for your effort. Here the log:

[errorlog.txt](https://github.com/user-attachments/files/19411177/errorlog.txt)

RoPoXx, 2025-03-25:
Hello, I have the same problem using the new version 8.1.1, I am attaching the logs to help. One detail, on my personal machine with Windows 11 it works normally. However, on my work machines with Windows 10 the error occurs. Using Naps2 v.7 it works normally on any Windows. If you need me to do any additional testing, I am available.

[debuglog.txt](https://github.com/user-attachments/files/19456696/debuglog.txt)
[errorlog.txt](https://github.com/user-attachments/files/19456697/errorlog.txt)

rmmbr, 2025-03-26:
> OCR no longer works for me in the portable version of NAPS2 8.1.1. In the portable version 7.5.3.0 this still worked. I am using Windows 10 64bit.
> 
> Here is the error message from the errorlog.txt:
> 
> 2025-03-22 13:19:47.2922 3472 Error running OCR System.IO.FileNotFoundException: Could not find file 'D:\NAPS\Data\temp\2l3qnth3.4gd.hocr'. File name: 'D:\NAPS\Data\temp\2l3qnth3.4gd.hocr' at SafeFileHandle Microsoft.Win32.SafeHandles.SafeFileHandle.CreateFile(string fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options) at SafeFileHandle Microsoft.Win32.SafeHandles.SafeFileHandle.Open(string fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options, long preallocationSize, UnixFileMode? unixCreateMode) at new System.IO.Strategies.OSFileStreamStrategy(string path, FileMode mode, FileAccess access, FileShare share, FileOptions options, long preallocationSize, UnixFileMode? unixCreateMode) at FileStreamStrategy System.IO.Strategies.FileStreamHelpers.ChooseStrategyCore(string path, FileMode mode, FileAccess access, FileShare share, FileOptions options, long preallocationSize, UnixFileMode? unixCreateMode) at new System.IO.FileStream(string path, FileMode mode, FileAccess access, FileShare share, int bufferSize) at Stream System.Xml.XmlDownloadManager.GetStream(Uri uri, ICredentials credentials, IWebProxy proxy) at void System.Xml.XmlTextReaderImpl.FinishInitUriString() at XmlReader System.Xml.XmlReaderSettings.CreateReader(string inputUri, XmlParserContext inputContext) at XDocument System.Xml.Linq.XDocument.Load(string uri, LoadOptions options) at async Task NAPS2.Ocr.TesseractOcrEngine.ProcessImage(ScanningContext scanningContext, string imagePath, OcrParams ocrParams, CancellationToken cancelToken)
> 
> If I watch the D:\NAPS\Data\temp\ folder I see that it creates a .jpg-file after a scan but no .hocr-file.

Confirmed on my machine: Windows 11 24H2 and NAPS 8.1.1. Exactly the same stacktrace. Normal bitmap-scanning is ok.

After reverting to NAPS 7.5.3 all problems are gone.

rmmbr, 2025-03-26:
The title of the issue should be changed to "OCR no longer works in <s>the portable</s> version 8.1.1", because the problem exists in the normal 64-bit-version too ...

cyanfish, 2025-03-26:
Can you try installing (or reinstalling) the [Microsoft VC++ Redist](https://aka.ms/vs/17/release/vc_redist.x64.exe)?

cyanfish, 2025-03-26:
And if that doesn't work, can you check which [version of Windows 10](https://learn.microsoft.com/en-us/windows/client-management/client-tools/windows-version-search#system-properties) you have?

TheBastelWastl, 2025-03-26:
> Can you try installing (or reinstalling) the [Microsoft VC++ Redist](https://aka.ms/vs/17/release/vc_redist.x64.exe)?

This worked for me! OCR runs without issues and the resulting PDF is seachable.
Many thanks for your support!

cyanfish, 2025-03-26:
@rmmbr Did you install using the MSI or EXE installer? If the [redist](https://aka.ms/vs/17/release/vc_redist.x64.exe) fixes it then using the MSI could be the issue, as the EXE will install it for you but the MSI/portable wont.

RoPoXx, 2025-03-26:
I have many VMs available here, I'll do some tests, using MSI and EXE.

RoPoXx, 2025-03-26:
Installing VCRedist on a machine that had the problem previously, reported here previously, started working correctly.

Now let's go to the tests:
VM602 - W10x64 22h2 Build **19045.5487** // VCRedist x64 14.32.**31326**.0 // NAPS2 x64 v.8.1.1 **MSI**
Test #1 > Installed Windows, > Installed NAPS2 **MSI**, > Run OCR > Result: **ERROR**
Test #2 > Installed VCRedist x64 **14.42.34438** > Reboot PC > Run OCR > Result: **WORKED**

VM603 - W10x64 22h2 Build **19045.5608** // VCRedist x64 14.32.31326.0 // NAPS2 x64 v.8.1.1 **MSI**
Test #1 > Installed Windows, > Installed NAPS2 **MSI** > Run OCR > Result: **ERROR**
Test #2 > Installed VCRedist x64 **14.42.34438** > Run OCR without rebooting > Result: **WORKED**

VM604 - W10x64 22h2 Build **19045.5487** // VCRedist x64 14.32.31326.0 // NAPS2 x64 v.8.1.1 **EXE**
Test #1 > Installed Windows, > Installed NAPS2 **EXE** > Ran OCR > Result: **ERROR**
Test #2 > Installed VCRedist x64 **14.42.34438** > Run OCR without rebooting > Result: **WORKED**

VM606 - W10x64 22h2 Build **19045.5608** // VCRedist x64 14.32.31326.0 // NAPS2 x64 v.8.1.1 **EXE**
Test #1 > Installed Windows, > Installed NAPS2 **EXE** > Run OCR > Result: **ERROR**
Test #2 > Installed VCRedist x64 **14.42.34438** > Run OCR without restarting > Result: **WORKED**

In summary: In all tests, using EXE or MSI, the error is the same. It is necessary to update the VCRedist in all of them.
It may be necessary to include the VCRedist in the NAPS2 setup

cyanfish, 2025-03-26:
@RoPoXx With the exe installer did you see the popup for it installing the redist?

RoPoXx, 2025-03-26:
No, no popups and no warnings.

cyanfish, 2025-03-26:
Thanks, I understand the issue now. The problem is that it needs a newer version of the redist to work, but the NAPS2 installer only requires an older version (and won't try and update if you already have that older version installed). I'll update the installer to require the newer version and/or see if I can fix the OCR so it doesn't need that newer version (which would be more helpful for the zip/msi).

rmmbr, 2025-03-27:
> [@rmmbr](https://github.com/rmmbr) Did you install using the MSI or EXE installer? If the [redist](https://aka.ms/vs/17/release/vc_redist.x64.exe) fixes it then using the MSI could be the issue, as the EXE will install it for you but the MSI/portable wont.

I installed naps2-8.1.1-win-x64.exe. On my Windows 11 Pro 24H2 (Build 26100.3476) system is redist 14.34.31931 installed. At the moment I stay with 7.5.3 because of lack of time ... I try later. Thanks for your take care of the problem ...

cyanfish, 2025-03-29:
This should be fixed in [8.1.2](https://www.naps2.com/download). I've bundled the relevant files with NAPS2 so you should no longer need the redist to be installed at all.

closed 2025-03-29 by cyanfish