cyanfish/naps2 #581: terrible scan quality

tadabrowski, 2025-03-23:
I’m experiencing a significant drop in scan quality compared to the manufacturer’s software (Brother iPrint&Scan). I’m using the same settings in both tools. In NAPS2, I disable all compression, yet the documents look terrible (a lot of artifacts around text) and have a much larger file size. 

I'm using ESCL protocol. I've tested all advanced settings like WIA version and TWAIN implementation without any significant effect.

**naps2**
[image]

**brother**
[image]

Windows 11
NAPS2 8.1.1.0
BROTHER Ink Benefit Plus DCP-T426W

zwo1a, 2025-06-09:
I also use a Brother printer and have encountered the same issue. It appears to be a limitation of Brother devices—when using wireless "universal protocols" such as ESCL or WIA, the printer only transmits heavily compressed images. This behavior is consistent across other applications as well. Moreover, the printer's web interface offers no settings to adjust image quality, and even selecting the highest quality in NAPS2 has no effect.

I tried using VueScan and noticed that when the printer is discovered via mDNS, the transmitted image quality is excellent. However, when using ESCL, the quality significantly deteriorates.

tadabrowski, 2025-09-08:
@christianclavet 
The lines visible on the scan are not a problem. They also appear on the scanned document. Mainly, I mean the typical compression artifacts visible i.e. around the letters.