cyanfish/naps2 #582: Imported pdf file show a black preview page.

Riaulo, 2025-03-24:
Since version 8 (even the beta) when I import a pdf file the preview of all pages is black. just enlarge or reduce the size of the displayed pages to make the correct preview appear. The attached screenshots refer to the version naps2-8.1.1-win-x64.

[image]
[image]

cyanfish, 2025-03-29:
What version of Windows are you using? Can you attach a sample PDF with the issue? Does it happen with all PDFs? In particular, does it matter if it's (A) a PDF from generated by scanning & saving from NAPS2 vs (B) a PDF from another source? Does it happen if you import an image (JPEG/PNG) file?

Riaulo, 2025-03-29:
What version of Windows are you using?
Edizione	Windows 11 Pro Insider Preview
Versione	Dev
Data installazione:	‎28/‎03/‎2025
Data di scadenza della valutazione	‎15/‎09/‎2025 23:16
Build sistema operativo	27823.1000
Feature Pack	Pacchetto di esperienze per funzionalità Windows 1000.26100.64.0
But it did this with previous versions too.

 Can you attach a sample PDF with the issue?
Yes

Does it happen with all PDFs?
Yes

 In particular, does it matter if it's (A) a PDF from generated by scanning & saving from NAPS2 vs (B) a PDF from another source?
Always. Even the scanned pages come out black.

Does it happen if you import an image (JPEG/PNG) file?
Yes

[Guida pratica del gatto.pdf](https://github.com/user-attachments/files/19516646/Guida.pratica.del.gatto.pdf)
[image]

Riaulo, 2025-04-02:
Hi, the program works correctly if you set the compatibility with windows 7, I hope it can help.

[image]

closed 2025-04-02 by Riaulo