cyanfish/naps2 #593: Force light mode (or dark mode)

vivadavid, 2025-04-03:
Hi,

I use Windows 11 with the dark theme, but I normally prefer having my programmes in light mode. Would it be possible to allow users to force light mode (or dark mode) regardless the theme they are using in Windows?

Thanks in advance.

githubuser181226, 2025-04-05:
IMO this shouldn't be forced, there should be an option to disable automatic theme switching and choose your own (light or dark) theme per desire.
I personally don't like the stark contrast of bright white scanned pages and black UI. This strains my eyes.
Please, disable the automatic light/dark mode and allow users the option to choose the theme.
Thank you.

cyanfish, 2025-04-08:
NAPS2 [8.1.4](https://www.naps2.com/download) now has a "Theme" setting.

closed 2025-04-08 by cyanfish

vivadavid, 2025-04-09:
I've just tested it and it works as expected. Thanks! :-)