cyanfish/naps2 #595: Clicking the "Save images" button fails to bring up file picker, does not save image

GitHubRub, 2025-04-06:
**Describe the bug**
Clicking the "Save images" button fails to bring up file picker. The button is highlighted blue (as if holding left-click) until the cursor is moved away from it. After clicking the button, exiting the program fails to end the process /usr/bin/naps2

**To Reproduce**
1. Scan a document
2. Click "Save images"

**Expected behavior**
Program brings up the file picker (either GTK or KDE, depending on naps2 version) in order for the user to pick the image's location.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Kubuntu 24.04
 - naps2 version: 8.1.3

**Additional context**
This problem started with version 7.2.0 and continues to exist in version 8.1.3. The problem is not present in version 7.1.2 (which coincidentally is the last version to use the GTK file picker by default even when running in a Plasma environment). Clicking "Save PDF" works fine, so there's some kind of asymmetry between the two buttons. Both the .deb and .flatpak installations exhibit the bug. Enabling "Image settings" → "Skip save prompt" causes the image to be saved but (obviously) prevents the user from specifying a location.

cyanfish, 2025-04-06:
Can you follow [these instructions](https://www.naps2.com/doc/troubleshooting#error-log) to collect error logs?

GitHubRub, 2025-04-08:
> Can you follow [these instructions](https://www.naps2.com/doc/troubleshooting#error-log) to collect error logs?

Unfortunately no log entries are generated in ~/.config/naps2/errorlog.txt when I click the "Save images" button, nor when I force close the process. errorlog.txt _does_ exist, but it only contains old entries pertaining to an old scanner driver I'm no longer using.

ProgMetalMan, 2025-04-15:
I am having the same problem with both the Save PDF button and the Save Images button on 8.1.4, using the flatpak and MATE on Void Linux. No error logs here either.

pablogila, 2025-05-20:
Hi there! I'm experiencing the same problem. I'm trying to export the images as @GitHubRub suggested, by enabling the "Skip save prompt". However, this exports the images as JPG by default. Is there any workaround to export them as TIFF, until this bug gets fixed? Thanks!

ProgMetalMan, 2025-10-07:
I finally got around to installing the 8.2.1.0 flatpak and I'm still having the problem on MATE.