cyanfish/naps2 #599: [Bug] Scan results in Duplex, even when Duplex not enabled

kirk56k, 2025-04-11:
Bug presents in:
- NAPS2 Windows Client
- NAPS2 SDK for .NET

OS: Windows 10 Enterprise Build 19045
Code Environment: .NET 9 WinForms
Scanner Driver: TWAIN

Expected Result: Scan with Feeder, get a single image per page. Scan with Duplex, get two images per page.

Actual Result: Scan with Feeder or Duplex, always get two images per page.

I have a Canon P-215II Desktop Scanner I'm using for testing. I am currently using your SDK, but I verified this happens in the full NAPS2 Windows Desktop client as well. If I set the Paper Source to either Feeder or Duplex, I get duplex either way (two images for one page). I know this isn't an issue with the scanner itself, as this problem does not present when using the bundled Canon Scanner software, or when using the NTwain library within a .NET 9 x86 Application or when using the TWAIN driver from Adobe Acrobat or other TWAIN enabled applications.

If you need any more details, please let me know. Thanks!

cyanfish, 2025-04-12:
Can you provide sample code with NAPS2.SDK and NTwain that produces different results?

cyanfish, 2025-04-12:
You can also play around with [TwainOptions](https://www.naps2.com/sdk/doc/api/NAPS2.Scan.TwainOptions.html) (e.g. try TwainDsm.Old and TwainTransferMode.Native).

kirk56k, 2025-04-12:
Hi Cyanfish,

Thanks for the quick response! I did play around with the TwainOptions within the NAPS2 Windows client, and it made no difference. I haven't tried changing that option within the SDK itself yet. As for a code sample, I can probably put something together on Monday or Tuesday for you. I was using NTwain directly as of just a few months ago, but I has having a hard time getting the PDF generation with Tesseract correct. My boss asked me to stop re-inventing the wheel though (something of a hobby for me, I already have too many libraries to support), so I found your project and began porting to it. Everything has worked great other than this Duplex issue.

kirk56k, 2025-04-14:
Well! That was quite the journey into TWAIN that I certainly wasn't anticipating today. So I found the issue. It turns out that its has to do with an odd way that the Canon TWAIN driver handles certain capabilities. The driver believes that "ExcludeBlankPages", "SimpleFeeder" and "DuplexFeeder" are all choices for a single option. Therefore, if you have "ExcluldeBlankPages" enabled or left to its default (which is enabled), it will use Duplex scanning regardless of what the Cap for Duplex is set to. In order to get proper Simplex Feeder scanning, you have to set Feeder enabled, Duplex disabled, and most importantly, ExcludeBlankPages to disable. I found this first by using the TWAIN source directly in a C++ app using twaindsm.dll directly. Then later with NTwain, and finally with your SDK. In all cases, setting the proper options works. Oddly, apparently, Adobe Acrobat does this already by default (there was no option for it that I could find, but it does correct Simplex scanning anyway with this source). So, sorry for the Odyssey, but the solution is found, and it's absolutely no fault of yours. I do very much appreciate the quick reply, though. Thanks for making such a thorough and well maintained SDK for scanning. There aren't many out there, proprietary or otherwise, and I'd always rather use something that isn't a black box. Thanks!

closed 2025-04-14 by kirk56k