cyanfish/naps2 #600: The application freezes if you try to split an image during background scanning

SeiOkami, 2025-04-12:
**Describe the bug**
A background scan of the next page is taking place. I'm trying to edit an existing one. I press the "Split" button and the application freezes.

**To Reproduce**
Steps to reproduce the behavior:
1. Отсканировать одну страницу
2. Запустить сканирование второй страницы в фоновом режиме
3. Пока идет сканирование, открыть первую страницу и нажать кнопку "Разделить"

**Expected behavior**
The application should not freeze.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 11 Pro]
 - Version 24H2

snachodog, 2025-04-16:
This might be a separate issue, but it sounds similar to what I'm experiencing - Saving a PDF and scanning in the background causes the application to freeze.

I am running Win10 Pro 22H2

DozeyM, 2025-04-21:
Yes, also hangs sometimes when using the crop tool

cyanfish, 2025-04-22:
What scanner? What driver type (twain/wia)? When you say freeze/hang, is that permanent or does it resolve itself once the scan finishes? If you scan from the Profiles window (then close the window and Split) does that freeze too? Is it fully consistent or only happens sometimes?

cyanfish, 2025-06-03:
Thanks, can you try this [test version](https://1drv.ms/u/c/2918c6d8dc4b6479/EU_vtrf9eLhOrsEtIUn_qRABnMerL5xoKEkc0ZGF8Adc1Q?e=4ylPQ2) and see if it fixes the issue?

TobiasKnauss, 2025-06-03:
I have tested it, the problem still exists, It freezes in the same way.
Are you not able to reproduce it yourself?

cyanfish, 2025-06-03:
Yeah, it works fine for me (though I have some more scanners in my closet I could pull out to test with). It's very strange that the test version didn't fix it, as it makes it so WIA scanning runs in an entirely separate process.

At what point exactly does it freeze? When you first open the crop window? A bit after the crop window opens? If you're dragging around the crop handles? If you press the OK button?

TobiasKnauss, 2025-06-03:
> At what point exactly does it freeze? When you first open the crop window? A bit after the crop window opens? If you're dragging around the crop handles? If you press the OK button?

It freezes in the moment I press the crop button in the edit window. The crop window never opens. I think the main thread is somehow blocked to open the cropping window, and tries over and over by starting new threads. I bet this is completely unrelated to WIA, but rather an internal issue in the UI handling, maybe because the update of the scan process bar takes place while the crop button is clicked. I scan at 300dpi which causes a quite fast update of the progress bar.

githubuser181226, 2026-01-11:
Hello.
In version 8.2.1.0 same exact issue is still there. I use Windows 10 x64, Brother devices via TWAIN.
The devices end the scanning but the program clearly stops receiving the image, because after killing it and restarting said image isn't restored.