cyanfish/naps2 #601: Native support for Windows 11 ARM64 - Snapdragon CPU

martinsuchan, 2025-04-13:
Feature description:
Few months ago I bought a Lenovo Yoga Slim 7x Copilot+ PC with Snapdragon X Elite CPU as my new primary PC, running Windows 11 ARM.

It would be great to have native ARM64 build of NAPS2 for Windows, right now I'm not even able to install the x64 package on my machine. I'm getting this error when launching the .exe installer:

[image]

Or at least the x64 version should run in emulated mode like tens of other apps.

Info about Windows 11 ARM64
https://support.microsoft.com/en-us/windows/windows-arm-based-pcs-faq-477f51df-2e3b-f68f-31b0-06f5e4f8ebb5

cyanfish, 2025-04-14:
You can probably run the [portable zip](https://www.naps2.com/download) even if the installer doesn't want to install.

martinsuchan, 2025-04-14:
I tested the portable version and it works. It is still strange and unintuitive that I need to use this version though.
Fixing the installer to have it run on on ARM should not be a problem, right?

Also looks like the app uses .NET Framework - the latest version 4.8.1 has native support form ARM64, it's worth a try.

cyanfish, 2025-04-14:
#567 is the tracking issue for this.