cyanfish/naps2 #602: Dialog buttons partially offscreen

bgoodmansf, 2025-04-18:
**Describe the bug**
The DELETE button is partially offscreen in the CustomPageSize dialog.

**To Reproduce**
1. Open New Profile dialog
2. Select "Custom" under Page Size
 
**Expected behavior**
Display DELETE button.

**Screenshots**
Attached

**Desktop (please complete the following information):**
 - OS: Fedora 41 with KDE desktop environment

IvarSnaaijer, 2026-10-20:
sorry for that, it decided that i did not need to see the change before committing... 
But after the second change it looks like this:

[image]

the name field is smaller, but seems adequate.

cyanfish, 2026-10-25:
Fixed in [8.6.0](https://www.naps2.com/download).

closed 2026-10-25 by cyanfish