cyanfish/naps2 #606: Scanned image shown in preview screen has bottom edge cut off

rogerval, 2025-04-22:
**Describe the bug**
A displayed scanned item in the preview window has the bottom edge hidden; clicking the zoom out button shows the bottom edge. Clicking the "scale with window" button shows the item with the bottom edge hidden again.

**To Reproduce**
Steps to reproduce the behavior:
1. Scan an item
2. Double-click the scanned image in the NAPS2 "home" screen.
3. Preview window appears with the scanned image's bottom edge hidden.

**Expected behavior**
Preview window should appear with the scanned image's bottom edge visible.

**Screenshots**
Please see screenshots of the problem; the first shows the problem, the second shows what happens when the zoom-out button is clicked.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 24H2 OS Build 26100.3775

**Additional context**
(Nothing in particular; I'm more than happy to provide further info if required.)

[image]

[image]

rogerval, 2025-05-20:
Forgot to mention: the software version is 8.1.4.0

RoPoXx, 2025-06-02:
Contributing, I have the same issue on this version (8.1.4.0).

[image]

CrazyTim71, 2025-08-21:
Sadly this bug is still present in 8.2.0.0.
I've spent some time investigating the code and believe I've identified the root cause.

The issue appears to be in ``ScrollZoomImageViewer.cs``, specifically in the ``AvailableHeight`` calculation:

```cs
private int AvailableHeight => _scrollable.Height - 2 - BorderOffset;

// ...

public void ZoomToContainer()
{
   // ...
    else
    {
        RenderSize = new Size((int) Math.Round(Image!.Width * AvailableHeight / (float) Image.Height), AvailableHeight);
        _renderFactor = AvailableHeight / (float) Image.Height;
    }
    // ...
}
```

The problem is that ``_scrollable`` returns the wrong height (e.g., 600px -> full window height), so ``AvailableHeight`` uses the full window height. I guess this happens because ``ZoomToContainer()`` gets called before the toolbar is fully rendered. So the image gets sized for the entire window, but then when the toolbar shows up and takes space, the image is too big and the bottom gets cut off.

I fixed it by subtracting the toolbar height (~50px) from ``AvailableHeight``. It works, but it's not a great solution because I had to hardcode the toolbar size (there's no way to get it dynamically as far as I know). This also creates extra empty space above the image because of the larger ``YOffset``.

```cs
private int AvailableHeight => _scrollable.Height - 2 - BorderOffset - 50;
```

@cyanfish I would really appreciate if you could take a closer look at this bug and give me some feedback. I'd be happy to work on a proper solution with some guidance. It has definitely catched my interest 😄

closed 2025-08-31 by cyanfish

rogerval, 2025-09-01:
Thanks cyanfish, your fix is much appreciated.