cyanfish/naps2 #607: Scan always aborts with no error apparent

Atario, 2025-04-23:
**Describe the bug**
Scan starts, gets partway through, then stops in the middle.  Scanner returns to rest position.  No page is shown and no error pops up.

**To Reproduce**
Steps to reproduce the behavior:
1. Attempt any page scan
2. Watch progress popup and listen to scanner whirring through part of the page
3. It stops early
4. The progress popup proceeds for a little while longer, then goes away
5. No page image is shown and there's no complaint about anything from NAPS2

**Expected behavior**
A page to be added in the main window

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Windows 11 Pro
 - NAPS2 Version 8.1.4.0

**Additional context**
Output in `errorlog.txt`:

````
2025-04-23 04:45:16.0664 32364 Error in invoked action System.NullReferenceException: Object reference not set to an instance of an object.
   at ReturnCode NTwain.Triplets.PendingXfers.EndXfer(TWPendingXfers pendingXfers)
   at void NTwain.Internals.TransferLogic.DoTransferRoutine(ITwainSessionInternal session)
   at void NTwain.TwainSession.HandleSourceMsg(Message msg)
   at ReturnCode NTwain.TwainSession.HandleCallback(TWIdentity origin, TWIdentity destination, DataGroups dg, DataArgumentType dat, Message msg, IntPtr data)+() => { } [1]
   at void NAPS2.Platform.Windows.Win32MessagePump.RunQueuedActions()
````

Output in `debuglog.txt`:
````
2025-04-23 04:43:51.1361 37776 Scanning with ScanDevice { Driver = Twain, ID = Kyocera MA2000w (Net) (1), Name = Kyocera MA2000w (Net) (1), IconUri =  } 
2025-04-23 04:43:51.1361 37776 Scan source: Flatbed; bit depth: Color; dpi: 600; page size: 8.5x11 in 
2025-04-23 04:43:51.2806 32364 Debug: Using new dsm. 
2025-04-23 04:43:51.3268 32364 Using TWAIN DSM: C:\Program Files\NAPS2\lib\_win32\twaindsm.dll 
2025-04-23 04:43:51.3268 32364 NAPS2.TW - Opening session 
2025-04-23 04:43:51.3268 32364 Debug: Thread 1: OpenManager. 
2025-04-23 04:43:51.3421 32364 NAPS2.TW - StateChanged (to 3) 
2025-04-23 04:43:51.3421 32364 Debug: Using TWAIN2 memory functions. 
2025-04-23 04:43:51.3421 32364 NAPS2.TW - Finding source 
2025-04-23 04:43:51.3421 32364 Debug: Source id = 1 
2025-04-23 04:43:51.3421 32364 NAPS2.TW - Opening source 
2025-04-23 04:43:51.3572 32364 NAPS2.TW - Name: Kyocera MA2000w (Net) (1); Manu: TWAIN Working Group; Family: KYOCERA CORPORATION; Version: 3.0 Version 2.000; Protocol: 1.8 
2025-04-23 04:43:51.3572 32364 Debug: Thread 1: OpenSource. 
2025-04-23 04:43:51.3572 32364 NAPS2.TW - StateChanged (to 4) 
2025-04-23 04:43:51.3572 32364 Debug: Registered callback2 OK. 
2025-04-23 04:43:51.3572 32364 NAPS2.TW - Configuring source 
2025-04-23 04:43:51.3572 32364 Defaulting to Memory transfer mode. 
2025-04-23 04:43:51.3572 32364 Transfer mode: Memory 
2025-04-23 04:43:51.3849 32364 NAPS2.TW - Enabling source 
2025-04-23 04:43:51.3849 32364 Debug: Thread 1: EnableSource with NoUI. 
2025-04-23 04:43:53.1024 32364 NAPS2.TW - StateChanged (to 5) 
2025-04-23 04:45:11.8677 32364 Debug: Thread 1: CallbackHandler at state 5 with MSG=XferReady. 
2025-04-23 04:45:11.8702 32364 Debug: Got TWAIN msg XferReady 
2025-04-23 04:45:11.8702 32364 NAPS2.TW - StateChanged (to 6) 
2025-04-23 04:45:15.4689 32364 Debug: Trying to raise event TransferReadyEventArgs on thread 1 without sync. 
2025-04-23 04:45:15.4689 32364 NAPS2.TW - TransferReady 
2025-04-23 04:45:15.4775 32364 Debug: Trying to raise event TransferErrorEventArgs on thread 1 without sync. 
2025-04-23 04:45:15.4775 32364 NAPS2.TW - TransferError 
2025-04-23 04:45:15.4775 32364 NAPS2.TW - Finishing with error NAPS2.Scan.Exceptions.AlreadyHandledDriverException: Exception of type 'NAPS2.Scan.Exceptions.AlreadyHandledDriverException' was thrown.
2025-04-23 04:45:15.4775 32364 Debug: Thread 1: ForceStepDown. 
2025-04-23 04:45:15.9942 32364 Debug: Thread 1: DisableSource. 
2025-04-23 04:45:16.0664 32364 Debug: Thread 1: CallbackHandler at state 6 with MSG=CloseDSReq. 
2025-04-23 04:45:16.0664 32364 NAPS2.TW - StateChanged (to 4) 
2025-04-23 04:45:16.0664 32364 NAPS2.TW - SourceDisabled 
2025-04-23 04:45:16.0664 32364 Debug: Thread 1: CloseSource. 
2025-04-23 04:45:16.0664 32364 NAPS2.TW - StateChanged (to 3) 
2025-04-23 04:45:16.0664 32364 Debug: Thread 1: CloseManager. 
2025-04-23 04:45:16.0664 32364 NAPS2.TW - StateChanged (to 2) 
2025-04-23 04:45:16.0664 32364 NAPS2.TW - StateChanged (to 6) 
2025-04-23 04:45:16.0664 32364 Error in invoked action System.NullReferenceException: Object reference not set to an instance of an object.
   at ReturnCode NTwain.Triplets.PendingXfers.EndXfer(TWPendingXfers pendingXfers)
   at void NTwain.Internals.TransferLogic.DoTransferRoutine(ITwainSessionInternal session)
   at void NTwain.TwainSession.HandleSourceMsg(Message msg)
   at ReturnCode NTwain.TwainSession.HandleCallback(TWIdentity origin, TWIdentity destination, DataGroups dg, DataArgumentType dat, Message msg, IntPtr data)+() => { } [1]
   at void NAPS2.Platform.Windows.Win32MessagePump.RunQueuedActions()
2025-04-23 04:45:16.0981 32364 Debug: Got TWAIN msg CloseDSReq 
2025-04-23 04:45:16.0981 32364 NAPS2.TW - Finishing with completion 
2025-04-23 04:45:16.0981 32364 Debug: Thread 1: ForceStepDown. 
2025-04-23 04:46:11.1292 31552 Debug: Using new dsm. 
2025-04-23 04:46:11.1798 31552 Debug: Thread 1: OpenManager. 
2025-04-23 04:46:11.1906 31552 Debug: Using TWAIN2 memory functions. 
2025-04-23 04:46:11.1906 31552 Debug: Source id = 1 
2025-04-23 04:46:11.1906 31552 Debug: Thread 1: CloseManager. 
2025-04-23 04:48:49.6767 37972 Debug: Using new dsm. 
2025-04-23 04:48:49.7260 37972 Debug: Thread 1: OpenManager. 
2025-04-23 04:48:49.7366 37972 Debug: Using TWAIN2 memory functions. 
2025-04-23 04:48:49.7366 37972 Debug: Source id = 1 
2025-04-23 04:48:49.7366 37972 Debug: Thread 1: CloseManager. 
2025-04-23 04:48:51.9674 36080 Debug: Using new dsm. 
2025-04-23 04:48:52.0148 36080 Debug: Thread 1: OpenManager. 
2025-04-23 04:48:52.0253 36080 Debug: Using TWAIN2 memory functions. 
2025-04-23 04:48:52.0253 36080 Debug: Source id = 1 
2025-04-23 04:48:52.0253 36080 Debug: Thread 1: CloseManager. 
2025-04-23 04:48:53.5130 38648 Debug: Using new dsm. 
2025-04-23 04:48:53.5597 38648 Debug: Thread 1: OpenManager. 
2025-04-23 04:48:53.5597 38648 Debug: Using TWAIN2 memory functions. 
2025-04-23 04:48:53.5723 38648 Debug: Source id = 1 
2025-04-23 04:48:53.5723 38648 Debug: Thread 1: CloseManager. 
2025-04-23 04:48:55.0826 21200 Debug: Using new dsm. 
2025-04-23 04:48:55.1315 21200 Debug: Thread 1: OpenManager. 
2025-04-23 04:48:55.1421 21200 Debug: Using TWAIN2 memory functions. 
2025-04-23 04:48:55.1421 21200 Debug: Source id = 1 
2025-04-23 04:48:55.1421 21200 Debug: Thread 1: CloseManager. 
````

cyanfish, 2025-04-24:
In your profile settings, under Advanced, Twain Implementation, if you select "Native Transfer" does it make a difference?

Atario, 2025-04-24:
Nope, same result.