cyanfish/naps2 #609: GetDeviceList for TWAIN crashes

rokoduran, 2025-04-25:
**Describe the bug**
Entire app freezes when attempting to fetch devices for TWAIN

**To Reproduce**
Steps to reproduce the behavior:
I am using an app where after a message via websockets I attempt fetching all available TWAIN sources. On the getDeviceList call the entire program freezes and just skips the rest of the function's execution.

 Before NAPS I used DTWAIN and I had similar issues - I needed to wrap TWAIN executions in STAThreads. I tried doing the same thing here but to no avail. Please help me out here, I'm losing it, there's nothing in the docs about these cases.

**Expected behavior**
Successfully fetches devices and the entire thread doesn't die

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10, macOS 12, Ubuntu 22.04]
 - Version [e.g. 7.1.0]

cyanfish, 2025-04-25:
What SDK version? What kind of app are you running (e.g WinForms desktop app, etc.).

rokoduran, 2025-04-25:
I'm using NAPS SDK version 1.1.6.  App is using WinForms, .Net Framework 4.8

cyanfish, 2026-10-25:
This should be fixed in SDK 1.6.0.

closed 2026-10-25 by cyanfish