cyanfish/naps2 #611: Execution always in Background

JPluess, 2025-05-01:
Currently, the scan execution progress appears in a separate window, which adds unnecessary clutter—there are already too many windows during scanning.

[image]

Instead, please consider always running the scan process in a minimized state, similar to the following example:

[image]

Or, at the very least, provide a setting to allow users to choose this behavior if you feel you can't detach from the current multi-window approach. And if you must keep the extra window, please make sure its title is clear and meaningful

cyanfish, 2025-05-09:
As far as the window title, it's the name of the scanner, which seems to be a [bug in recent versions of windows](https://www.google.com/search?q=wia+"no+friendly+name")...

Basty7, 2025-06-26:
> As far as the window title, it's the name of the scanner, which seems to be a [bug in recent versions of windows](https://www.google.com/search?q=wia+%22no+friendly+name%22)...

@cyanfish I believe you misunderstood the request:
I guess I am asking for the exact same thing which would be a toggle in the settings so that scanning does not produce this popup window, a behavior which we can get by pressing the "Run in background" button, but it will always default to re-generate the window. Being a tilingWM user this looks even worse than on windows.

So this issue should be a Feature request: "Add toggle to always run in background"

kuro68k, 2025-06-26:
Yes, the ideal scenario would be that I can just keep feeding the scanner and pressing a button for each page, without it interrupting the work I have on screen. Complete background operation.

I set it so that one of the buttons on my scanner runs NAPS2, which works fine except that every time I press it the window pops to the front.

antonkesy, 2026-09-22:
Especially annoying when using tiling window managers 💀