cyanfish/naps2 #613: Please add a  "Disable delete confirmation" setting.

triyanroy, 2025-05-02:
I edit big files and the confirmation prompt every single time is very irritating.

A setting to simple disable the delete confirmation prompt.

Could not find any alternative.

pexner, 2025-05-15:
I'd like to see a checkmark in the confirmation dialog named "Don't show this dialog again." and a setting in global preferences to "Reset warning dialogs". Thanks in advance.

loremari, 2026-10-27:
As a suggestion, confirmation prompts become unnecessary when an Undo action is available.

A great improvement in usability could be to completely remove the confirmation prompts after clicking Delete or Clear for everyone, and add an Undo button to reverse the action.

This would render the configuration setting asked here unnecessary.