cyanfish/naps2 #615: Duplex scanning

Lili68-T, 2025-05-07:
NAPS version 8.1.4 (naps2-8.1.4-linux-x64.deb distribution) on Astra Linux 1.7.5 OS (Debian 12 kernel) does not perform duplex scanning when using Kyocera 2040 MFP. Profile parameters are configured (screenshot). As a result, only the first pages of each sheet are scanned from ADF.

[image]

cyanfish, 2025-05-09:
Can you follow [these instructions](https://www.naps2.com/doc/troubleshooting#debug-log) to collect debug logs?

Lili68-T, 2025-05-12:
[debuglog.txt](https://github.com/user-attachments/files/20153873/debuglog.txt)

Lili68-T, 2025-05-15:
Is this information enough?

cyanfish, 2025-05-25:
It looks like Kyocera has non-standard options for the scanner source. If you use XSane, it should show the options as "Auto", "DP", "Platen", "2-sided (Binding Top)", "2-sided (Binding Left/Right)". Can you maybe try these options in XSane and see how each work?

My guess is that "Auto" uses the feeder if present and the flatbed otherwise. "Platen" should just be the flatbed. But between "DP" and the "2-sided" options I'm not sure which is the right choice for duplex scanning.