cyanfish/naps2 #616: Separate buttons for "Save All" and "Save Selected"

TobiasKnauss, 2025-05-10:
**Is your feature request related to a problem? Please describe.**
I often scan multiple single-page documents at once, and later I save them to different PDFs.  
For saving, I have to use the dropdown part of the Save to PDF button (I'm using the German GUI, so I don't know the exact names). If I click the wrong menu entry, then I accidentally save all scanned pages, which happens only sometimes, but it has happened.

**Describe the solution you'd like**
It would be nice to have "Save Selected" as default when one or multiple pages are selected. But this would break the default behaviour, so the change of this behaviour should be configurable in the options.
Even better would be having 2 separate buttons.

cyanfish, 2025-05-10:
If you click Settings you can change the default.

TobiasKnauss, 2025-05-10:
@cyanfish  Oh, there are settings?
Sorry, but that was not a joke. In the default size of the program, the setting button was hidden due to little space in the toolbar. Now I found it, and I found the option I was looking for. Thanks!

closed 2025-05-10 by TobiasKnauss