cyanfish/naps2 #620: macos - Scan to Apple Mail.app not working

pexner, 2025-05-15:
Set Email Settings to provider "Apple Mail".
Scan
Email All
Result "beep" and nothing happens.

**Expected behavior**
Mail.app opens, creates new mail and adds scan as attachment.

Tested on macOS 14 and 15.

cyanfish, 2025-05-15:
It works for me, but it sometimes creates the window for the new draft underneath the Inbox window.

Have you tried with the Mail app both open and closed?

pexner, 2025-05-18:
I found the issue.
If the default application for mail is NOT Mail.app (settings in Mail.app Preference -> General) it's not working.
My default aplication for mail is Outlook. But I use Mail.app for my personal mails.
When setting Mail.app as default app, it's working.
On Windows, naps2 offers more mail handling apps to use.

cyanfish, 2025-06-08:
I've had a look and unfortunately Apple doesn't really let apps talk to mail apps in the same way as Windows. I've made it so the Apple Mail option is disabled in [8.2.0](https://www.naps2.com/download) if it's not the default, so that people don't select it and then wonder why it's not working, but I don't think it's possible to have Mail.app work without it being the default.

cyanfish, 2025-06-10:
Fair. I'm not going to put in the effort to do that for each client but I'll leave open this issue in case someone wants to contribute a fix.