cyanfish/naps2 #622: Placeholders not working when Autosave enabled

e-rossman, 2025-05-19:
**Describe the bug**
Using placeholder $(nnn) in an profile with autosave enabled creates files with literal $(nnn) in the filename.
successfully and the autosave happens but the filename still has the `$(nnn)` literally in the filename. I deleted my hand-typed version and clicked the example in the placeholders to insert it (in case I was mistakenly putting an extraneous non-printable character in) but this did not help. I tried restarting NAPS2 without success.

**To Reproduce**
Steps to reproduce the behavior:
1. Click on Scan
2. Filename is saved without replacement of the placeholder

**Expected behavior**
I was expecting the placeholder to be replaced with the next sequential 3-digit number.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 8.1.4.0

**Additional context**
The same profile kind of works correctly when in batch mode. Since autosave is enabled, it saves the PDF twice. Once with the literal $(nnn) and once with the placeholder replaced.

cyanfish, 2025-05-19:
You're using the wrong type of bracket on one side, { vs (.

e-rossman, 2025-05-20:
Hmm, I didn't think I was using { anywhere. The image does make it look like it, so it's possible. I will check shortly.

e-rossman, 2025-05-20:
OK, consider me embarrassed. I was looking at the wrong profile when I saw the correct `$(nnn)`

thanks for dealing with my silly typo.

closed 2025-05-20 by e-rossman